Qrome/printer-monitor #3: Crash when I configure through the web page

Fienberber, 2018-05-20:
So as the tile says ... 
when I try to configure the Wemos through the web page the Wemos simply reset. 
I can't even change the source code because it doesn't change anything. 
It can be the fault of my Wemos (maybe somewhere broken), I wait a new one but I'm wondering if someone has the same issue. 

Thank for you project it's awesome btw !

wabbitguy, 2018-05-24:
For my own usages, I don't own any Wemos so I'm using a NodeMCU (12E) and I use the 2.3.0 ESP8266 boards config. It works perfectly as well (far more reliable for my projects for the NodeMCU than the 2.4.0 and 2.4.1 configs). Next on my list is using an NTP server instead of Google. BTW super slick way of doing the web page config!!!!!

Qrome, 2018-05-25:
@wabbitguy -- thank you.  It kind of fell into place.  I have a few other projects that helped polish the web page config.

Fienberber, 2018-05-25:
Ok i’ll Try it out ! 
I haven’t much time this week ... 
I think the broblem come from the wemos witch don’t work as expected.
Thanks for your great job !

closed 2018-05-25 by Fienberber