Qrome/printer-monitor #10: Print Finished

Steff1990, 2018-05-28:
I have Problems at the moment my printer finished. After that, time remainig displays an old value. Is it possible to fix it or give a message, finished?

wabbitguy, 2018-05-29:
The time remaining can be random as I've found, but the percentage shown at the bottom of the display always shows 100% when the print is completed.

wabbitguy, 2018-05-29:
Change the routine in printermonitor that prints the time remaining to fix the issue:

`void drawScreen2(OLEDDisplay *display, OLEDDisplayUiState* state, int16_t x, int16_t y) {
  display->setTextAlignment(TEXT_ALIGN_CENTER);
  display->setFont(ArialMT_Plain_16);
  display->drawString(64 + x, 0 + y, "Time Remaining");
  display->setFont(ArialMT_Plain_24);
  int val = printerClient.getProgressPrintTimeLeft().toInt();
  int days = elapsedDays(val);
  int hours = numberOfHours(val);
  int minutes = numberOfMinutes(val);
  int seconds = numberOfSeconds(val);
// this next section will zero the time remaining when the completion is 100%
  String checkPercentage = printerClient.getProgressCompletion();
  if (checkPercentage == "100") {// if we have printed 100%, zero out the time
    days = 0;
    hours = 0;
    minutes = 0;
    seconds = 0;
  }
// end of added code
  String time = zeroPad(hours) + ":" + zeroPad(minutes) + ":" + zeroPad(seconds);
  display->drawString(64 + x, 14 + y, time);
}`

Mel

Qrome, 2018-05-29:
Thanks guys.  I have put a fix in at the OctoPrintClient.cpp -- that way thing asking for the PrintTimeLeft will get "0" when the print is 100% -- this will be in 1.5 release.

closed 2018-05-29 by Qrome