Qrome/printer-monitor #8: OLED viewing orientation

wabbitguy, 2018-05-28:
I'm using NodeMCU 1.0's, but I had to add an option to rotate the display into its normal orientation.

Settings.h
boolean INVERT_DISPLAY = true;// flip the display to normal orientation, pins at top when viewing the face

printermonitor
  // initialize display
  display.init();
  if (INVERT_DISPLAY == true) display.flipScreenVertically();// connections at top of OLED display

  // Initial UI takes care of initializing the display too.
  ui.init();
  if (INVERT_DISPLAY == true) display.flipScreenVertically();//connections at top of OLED display

Qrome, 2018-05-28:
Yes, this is a good idea since others may be using different boxes / orientations for the software.  I wrote the code for my original box but now it supports a 1.3" OLED SSH1106 monitor as well and the option to flip the display is a good one.

Thanks.

wabbitguy, 2018-05-28:
I have several SH1106's in my parts bin, don't use them very often but when you stuck in the mods for that, I grabbed one and added the lines to make it work. OH YEAH! Fabulous! The little displays just went back into the bin...LOL

I'm designing a holder and case for the larger displays right now, already got the cases and holders for the 1306's done WAY NICER to look at!

[octomon](https://user-images.githubusercontent.com/8953419/40598125-befd1fd8-61fa-11e8-87db-ced49969c624.png)

Qrome, 2018-05-28:
Yeah, I just finished my box for the 1.3" SH1106.  I have not shared it yet, but I can send you the STL files if you want to have a look.

I have pushed the changes for the rotation of the display in the 1.5 dev branch.  Thanks again!

wabbitguy, 2018-05-28:
Popped you a PM via Thingiverse...

Mel

Qrome, 2018-05-28:
Great, I sent you a link to the case.  Thanks.

closed 2018-05-28 by Qrome