Qrome/printer-monitor #9: Temp display order...

wabbitguy, 2018-05-28:
Probably just my preference but on my LCD printer displays that I've been staring at for the last 3,700 hours (2004 LCD's Marlin) they always read tool temp/bed temp.

I added the orientation as an option:

Settings.h
boolean SHOW_TOOL_BED = true;// left right orientation of temp displays

printermonitor
void drawScreen1(OLEDDisplay *display, OLEDDisplayUiState* state, int16_t x, int16_t y) {
  display->setTextAlignment(TEXT_ALIGN_CENTER);
  display->setFont(ArialMT_Plain_16);
  int bed = printerClient.getTempBedActual().toInt();
  int tool = printerClient.getTempToolActual().toInt();
  if (SHOW_TOOL_BED == true) {// shows tool temp/bed temp
    display->drawString(64 + x, 0 + y, "Tool / Bed Temp");
    display->setTextAlignment(TEXT_ALIGN_LEFT);
    display->setFont(ArialMT_Plain_24);
    display->drawString(2 + x, 14 + y, String(tool));
    display->drawString(76 + x, 14 + y, String(bed));
    display->setFont(ArialMT_Plain_16);
    display->drawString(42 + x, 14 + y, "C");
    display->drawString(104 + x, 14 + y, "C");
  } else {
    display->drawString(64 + x, 0 + y, "Bed / Tool Temp");
    display->setTextAlignment(TEXT_ALIGN_LEFT);
    display->setFont(ArialMT_Plain_24);
    display->drawString(2 + x, 14 + y, String(bed));
    display->drawString(64 + x, 14 + y, String(tool));
    display->setFont(ArialMT_Plain_16);
    display->drawString(38 + x, 14 + y, "C");
    display->drawString(115 + x, 14 + y, "C");
  }
}

Mel

Qrome, 2018-05-28:
Thanks for the suggestion.  I am totally fine with Bed / Tool -- the other reason I am going to pass on this is that I don't want to clutter up the Settings.h with too many options that are minor in display.  Thanks!

closed 2018-05-28 by Qrome

wabbitguy, 2018-05-28:
No worries, merely a suggestion. I showed my modded version at the local maker space on Friday night and they really liked it and they preferred the tool/bed origination because all the printers at the space display the same way. I just stuck it in the Settings file because it's not something that really needs to be a user thing from the web interface. I.e. set in the code once, forget. Done.

I still plan on replacing the Google time, with a good old fashion NTP server call. Then add in the DST time rules for those that need it (like me...:-)

Mel