Qrome/printer-monitor #12: Not detecting offline state

wabbitguy, 2018-05-30:
When Printer-monitor detects Octoprint is operational, it changes from "clock" to the 3 sliding displays. If you do not do a print job, but power down Octoprint, printer monitor changes back to the "clock" mode.

However, if any print job has been started or ended and then Octoprint is shutdown, Printer Monitor remains in the operational state (sliding displays of the last print job).

In OctoprintClient.ccp, two lines added in the OctoPrintClient::getSubmitRequest will correct it.

The line is:
printerData.isPrinting = false;// so we aren't printing if the connection failed
Add it after nulling out the printerData.state="" lines

WiFiClient OctoPrintClient::getSubmitRequest(String apiGetData) {
  WiFiClient printClient;
  printClient.setTimeout(5000);

 Serial.println("Getting Octoprint Data");
  Serial.println(apiGetData);
  result = "";
  if (printClient.connect(myServer, myPort)) {  //starts client connection, checks for connection
    printClient.println(apiGetData);
    printClient.println("Host: " + String(myServer) + ":" + String(myPort));
    printClient.println("X-Api-Key: " + myApiKey);
    printClient.println("User-Agent: ArduinoWiFi/1.1");
    printClient.println("Connection: close");
    if (printClient.println() == 0) {
      Serial.println("Connection to " + String(myServer) + ":" + String(myPort) + " failed.");
      Serial.println();
      printerData.error = "Connection to " + String(myServer) + ":" + String(myPort) + " failed.";
      printerData.state = "";
      printerData.isPrinting = false;// if the connection failed, switch back to clock (not printing)
      return printClient;
    }
  } 
  else {
    Serial.println("Connection to OctoPrint failed: " + String(myServer) + ":" + String(myPort)); //error message if no client connect
    Serial.println();
    printerData.error = "Connection to OctoPrint failed: " + String(myServer) + ":" + String(myPort);
    printerData.state = "";
    printerData.isPrinting = false;// if the connection failed, switch back to clock (not printing)
    return printClient;
  }

Qrome, 2018-05-30:
I am adding a private method to the OctoPRintClient for resetting the printerData.   That way the web interface won't display the old data as well -- or anything else that would ask for it.

`// Reset all PrinterData
void OctoPrintClient::resetPrintData() {
  printerData.averagePrintTime = "";
  printerData.estimatedPrintTime = "";
  printerData.fileName = "";
  printerData.fileSize = "";
  printerData.lastPrintTime = "";
  printerData.progressCompletion = "";
  printerData.progressFilepos = "";
  printerData.progressPrintTime = "";
  printerData.progressPrintTimeLeft = "";
  printerData.state = "";
  printerData.toolTemp = "";
  printerData.toolTargetTemp = "";
  printerData.filamentLength = "";
  printerData.bedTemp = "";
  printerData.bedTargetTemp = "";
  printerData.isPrinting = false;
  printerData.error = "";
}`

This will be called  when a connection can not be made.  The error then will be updated into the object. 
This will be in the 1.5 branch.

Thanks!

Qrome, 2018-05-30:
I have committed my changes to the 1.5 branch here:  https://github.com/Qrome/printer-monitor/tree/1.5

If you can help me test this out, that would be great.   Thanks!

Qrome, 2018-05-30:
Yes, I do have a couple of printers.   I have tried to work out a display pattern that would work well with multiple printers.

wabbitguy, 2018-05-31:
I'm thinking a really big TFT display...:-)

wabbitguy, 2018-05-31:
So far in testing, the 1.5 it's working the way it should. Thus nulling all the data variables with Thor's hammer seems to have done the trick...I'll keep an eye on it over the next couple of days to make sure.

Qrome, 2018-05-31:
Great, thanks.  It seems to be working for me.   I had actually intended on making the reset method to blank out all the data, but never got around to it.  This was the perfect case of data sticking around when it is old and server offline.   I will close the issue here.   This will be part of 1.5 release.

closed 2018-05-31 by Qrome