Qrome/printer-monitor #13: Add basic auth(haproxy) support

eightheads, 2018-05-31:
Request to add basic auth support using haproxy. Currently if haproxy is in use with basic auth a HTTP/1.0 401 Unauthorized response is given.

Qrome, 2018-05-31:
Thanks, I will take a look at this later this week.

eightheads, 2018-05-31:
Thanks, let me know if you want me to test.

Qrome, 2018-05-31:
@eightheads -- Yes, please.   I am not able to test it, but I have coded for it and have it in a dev branch here:  https://github.com/Qrome/printer-monitor/tree/1.5BasicAuth

It should function -- I plan on adding the user / pass to the web ui as well.   Currently they are set in the Settings.h file.   If you can try it out, that would be great.

Qrome, 2018-05-31:
Hey, that is great news.  I will add the web ui portion so everything can be configured through the web interface and this will be part of the 1.5 release.
Thanks for testing.

Qrome, 2018-06-01:
The release 1.5 is out and final.  Thanks for the help testing.

closed 2018-06-01 by Qrome