Qrome/printer-monitor #17: 'DynamicJsonBuffer' error

c-ascha, 2018-06-08:
hey, have some problems while compiling.
do your have a solution?

thank you!

error messages:

Arduino: 1.8.5 (Linux), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

sketch/OctoPrintClient.cpp: In member function 'void OctoPrintClient::getPrinterJobResults()':
OctoPrintClient.cpp:125: error: 'DynamicJsonBuffer' was not declared in this scope
   DynamicJsonBuffer jsonBuffer(bufferSize);
   ^
sketch/OctoPrintClient.cpp:125:3: note: suggested alternative:
In file included from /home/sascha/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /home/sascha/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /home/sascha/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from sketch/OctoPrintClient.h:26,
                 from sketch/OctoPrintClient.cpp:24:
/home/sascha/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
OctoPrintClient.cpp:125: error: expected ';' before 'jsonBuffer'
   DynamicJsonBuffer jsonBuffer(bufferSize);
                     ^
OctoPrintClient.cpp:128: error: 'jsonBuffer' was not declared in this scope
   JsonObject& root = jsonBuffer.parseObject(printClient);
                      ^
OctoPrintClient.cpp:161: error: expected ';' before 'jsonBuffer2'
   DynamicJsonBuffer jsonBuffer2(bufferSize2);
                     ^
OctoPrintClient.cpp:164: error: 'jsonBuffer2' was not declared in this scope
   JsonObject& root2 = jsonBuffer2.parseObject(printClient);
                       ^
exit status 1
'DynamicJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Qrome, 2018-06-08:
Make sure you have the latest library files installed.  I am sure this is the issue.

akurz42, 2018-06-08:
I've got same when I updated to 6.0.0-beta, Rolling back to 5.13.1 should fix this.

Qrome, 2018-06-08:
Yeah,  noticed today that there was a 6.0.0-beta -- that is probably the issue.

Qrome, 2018-06-09:
I have confirmed that the inew 6.0.0-beta version of ArduinoJson has breaking changes in it.   It is noted in their release notes.   It will require anyone that uses it to make the changes in their code for it to work.

For now please use the stable 5.13.1 of ArduinoJson

c-ascha, 2018-06-15:
okay, thank you!!

closed 2018-06-15 by c-ascha