Qrome/printer-monitor #21: FR- Would it be possible to have it monitor 2 instances of octoprint, for different 3d printers?

monterman, 2018-06-15:
HI, would it be possible that PRINTER_MONITOR, check the same information form two different octorint instances.  
IT would require duplication of the OctoPrintClient.h and allow version OctoPrintClient1 and OctoPrintClient2 in configuration.   

This way we don't need one monitor per printer... 

Thanks!

Qrome, 2018-06-16:
I have been considering this, however it is much bigger than just what you are saying.   The display pattern and screens will need to be updated.   I am not saying this won't or can't be done.  It will require some thought and layout as well.  

One of the first things that will be done before this will happen is breaking out the configuration to multiple pages.   That is next on my list.
Thanks.

wabbitguy, 2018-06-17:
Personally I'd like multiple printers displayed at the same time but at the same time I'd want a much larger TFT for the display so each printer has it's own section to display in.

TheLion, 2019-08-16:
Any update on this enhancement? Second printer coming in shortly ;-)

doucettom, 2023-02-02:
I'm also wondering, I'd like it to shuffle through all actively printing machines :)