Qrome/printer-monitor #23: 24h problem

sciaio, 2018-06-17:
Why it shows AM/PM even in 24h mode? It is possible to show something else instead? I don't know current time or weather temperature. It's odd seeing 15:30 PM, very appreciated, thanks

wabbitguy, 2018-06-17:
Simple fix in printermonitor:

`
void drawClockHeaderOverlay(OLEDDisplay *display, OLEDDisplayUiState* state) {
  display->setColor(WHITE);
  display->setFont(ArialMT_Plain_16);
  if (!IS_24HOUR) {// hide the AM/PM on 24
    String time = timeClient.getAmPm();
    display->setTextAlignment(TEXT_ALIGN_LEFT);
    display->drawString(0, 48, time);
  }
  display->setFont(ArialMT_Plain_16);
  display->setTextAlignment(TEXT_ALIGN_CENTER);
  display->drawString(64, 48, "offline");
  display->setTextAlignment(TEXT_ALIGN_LEFT);

  display->drawRect(0, 43, 128, 2);

  drawRssi(display);
}
`

Qrome, 2018-06-18:
Yes, I am fully aware that in the 24 hour mode it still will display either AM or PM down below the line.  it is easy to remove that line as noted above.  Changing it to Weather Temp would be easy.

Qrome, 2018-06-18:
I have started the 2.1 dev branch here:  https://github.com/Qrome/printer-monitor/tree/2.1

I have updated it to display the host name where the AM/PM would show but only on 24Hour clock mode.  You can try out that branch now if you like.