Qrome/printer-monitor #29: Cant access /configure

asm7100, 2018-06-30:
When i trye to acess /configure from the menu, is says login and when i use the default admin/password it wait a littel and then ask for login again, and this dos i over and over again :-)

dataman, 2018-06-30:
Did you set the 2 user IDs and passwords in the settings.h file before you compiled? 

Sent from my iPhone 7!
Charley Jones, PMP

asm7100, 2018-06-30:
Hi.
This is what is in the settings.h

const int WEBSERVER_PORT = 80; // The port you can access this device on over HTTP
const boolean WEBSERVER_ENABLED = true;  // Device will provide a web interface via http://[ip]:[port]/
char* www_username = "admin";  // User account for the Web Interface
char* www_password = "password";  // Password for the Web Interface
float UtcOffset = -7; // Hour offset from GMT for your timezone
boolean IS_24HOUR = true;     // 23:00 millitary 24 hour clock
int minutesBetweenDataRefresh = 15;
boolean DISPLAYCLOCK = true;   // true = Show Clock when not printing / false = turn off display when not printing

asm7100, 2018-06-30:
Don't ask me way, sat www_username = "Admin"
And
www_password = "Password" (Big first letter)
Compiled and uploadet.
And now it works.????

closed 2018-06-30 by Qrome