Qrome/printer-monitor #33: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

firex726, 2018-07-14:
When compiling I get the following error:

`Arduino: 1.8.5 (Windows 7), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Users\Isaac\Desktop\printer-monitor-master\printermonitor\printermonitor.ino: In function 'void drawWeather(OLEDDisplay*, OLEDDisplayUiState*, int16_t, int16_t)':

printermonitor:808: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

   display->setFont((const uint8_t*)Meteocons_Plain_42);

                                                      ^

In file included from C:\Users\Isaac\Documents\Arduino\libraries\esp8266-oled-ssd1306-master/SH1106Wire.h:31:0,

                 from sketch\Settings.h:48,

                 from C:\Users\Isaac\Desktop\printer-monitor-master\printermonitor\printermonitor.ino:28:

C:\Users\Isaac\Documents\Arduino\libraries\esp8266-oled-ssd1306-master/OLEDDisplay.h:187:10: error:   initializing argument 1 of 'void OLEDDisplay::setFont(const char*)' [-fpermissive]

     void setFont(const char *fontData);

          ^

exit status 1
invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

labomat, 2018-07-14:
I fixed this by cleaning up my libraries: Now using only OLED_SSD1306_-_SH1106 by Stefan Frings, Fabio Durigon and have no errors anymore.

firex726, 2018-07-14:
Yep, that fixed it. 

Thanks!

closed 2018-07-14 by firex726