Qrome/printer-monitor #35: Web URL as connection?

BryanArnold, 2018-08-03:
I tried to use my duckdns address to connect to my octoprit so I can use my monitor at work. When I enter the duckdns address and save it, the config page reverts back to the internal ip of octoprit.

Is it possible to use a web URL to connect?

Qrome, 2018-08-03:
Remove the hostname value and leave it blank.   Then put in the duckdns address in where the IP is.   It should retain it.   If you provide the hostname field the device tries to do an IP lookup.   This should fix it for you.   Let me know how that goes.

BryanArnold, 2018-08-20:
Thanks. I finally had a chance to try. Removing the hostname did seem to make it further, but still did not connect. I am getting a Connection to OctoPrint failed message on the web interface. I can connect fine via the browser and the url. Maybe it's a HTTPS or auto trusting a self-signed certificate issue?