Qrome/printer-monitor #36: #define DISPLAY_SH1106

35nine, 2018-08-04:
Hi, I have this fantastic project up and running, one question for this 1.3 display.  The entire display is shifted down one pixel or line I guess is more accurate, the bottom row of the word 'offline' is missing.

Any way to edit some line in IDE to raise it up one line, am digging around but this is well above my head currently, am working on it.

Thank you very much

Qrome, 2018-08-05:
Odd -- I have not see that behavior on the 1.3 display using the SH1106 driver.  The code tells it to print absolute positions like 0,0 -- top left and so on.    Do you have a link to the actual display you bought?  Does it mention the use of the SH1106 driver or a different one?

35nine, 2018-08-05:
I hope this link is ok, posting this from a mobile:

https://www.ebay.com.au/itm/1-3-White-OLED-LCD-4PIN-Display-Module-IIC-I2C-Interface-128x64-for-Arduino/401395109681?hash=item5d75034f31:g:WH4AAOSwVGFZqEDx

Qrome, 2018-08-05:
The description says Controlling Chip:SSH1106 -- so that should work.   Can you share a photo of what the screen looks like?   From the description, it sounds like it should be working.

Qrome, 2018-08-05:
Also, I just have to check... in the Settings.h file -- you have removed the // at the start of this line:

`//#define DISPLAY_SH1106`

So that it looks like this:

`#define DISPLAY_SH1106`

I assume that you have since the title shows that.  I just wanted to make sure.

35nine, 2018-08-05:
Thank you for the reply, appreciate it.  Confirmed, line #define DISPLAY_SH1106 has no //.

The screen is hot glued into to the case, hindsight is a fun thing, I should have tested outside.  I'll do best to get a shot later today and maybe prise it back out