Qrome/printer-monitor #37: Compile Issue

AlexanderDNye, 2018-08-08:
When trying to compile I get this error. Any ideas?

`Arduino: 1.6.8 (Windows 10), Board: "WeMos D1 R1, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

G:\Documents\Arduino\printer-monitor-master\printer-monitor-master\printermonitor\printermonitor.ino: In function 'void setup()':

printermonitor:172: error: 'readSettings' was not declared in this scope

   readSettings();

                ^

printermonitor:256: error: 'displayPrinterStatus' was not declared in this scope

     server.on("/", displayPrinterStatus);

                    ^

printermonitor:257: error: 'handleSystemReset' was not declared in this scope

     server.on("/systemreset", handleSystemReset);

                               ^

printermonitor:258: error: 'handleWifiReset' was not declared in this scope

     server.on("/forgetwifi", handleWifiReset);

                              ^

printermonitor:259: error: 'handleUpdateConfig' was not declared in this scope

     server.on("/updateconfig", handleUpdateConfig);

                                ^

printermonitor:260: error: 'handleUpdateWeather' was not declared in this scope

     server.on("/updateweatherconfig", handleUpdateWeather);

                                       ^

printermonitor:261: error: 'handleConfigure' was not declared in this scope

     server.on("/configure", handleConfigure);

                             ^

printermonitor:262: error: 'handleWeatherConfigure' was not declared in this scope

     server.on("/configureweather", handleWeatherConfigure);

                                    ^

printermonitor:263: error: 'redirectHome' was not declared in this scope

     server.onNotFound(redirectHome);

                       ^

printermonitor:289: error: 'flashLED' was not declared in this scope

   flashLED(5, 500);

                  ^

printermonitor:290: error: 'findMDNS' was not declared in this scope

   findMDNS();  //go find Octoprint Server by the hostname

[... 234 lines cut ...]
printermonitor:1062: error: 'enableDisplay' was not declared in this scope

     enableDisplay(true);

                       ^

printermonitor:1074: error: 'enableDisplay' was not declared in this scope

     enableDisplay(false);

                        ^

printermonitor:1080: error: 'enableDisplay' was not declared in this scope

       enableDisplay(true);

                         ^

exit status 1
'readSettings' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Qrome, 2018-08-08:
Wow -- well... do you have all the files that go with the printermonitor.ino file in the same directory.   Do you have all the supporting library files installed and in the path.   

Please take a look at the detailed video done by Chris Riley.  It has lots of good information on making sure you have everything setup correctly to compile the code.   https://youtu.be/Rm-l1FSuJpI

wabbitguy, 2018-08-08:
Windows Arduino 1.6.8 is more than two years out of date as well (March 2016). A lot has changed since those days in the IDE as well as supporting libraries.

AlexanderDNye, 2018-08-08:
Lol wow, so I'll update my IDE and try again.

And that's actually the video I've been using.

Qrome, 2018-08-08:
Ahhh, good catch guys.   Let us know how it goes.

wabbitguy, 2018-08-08:
Yeah, been bitten myself a few times with a few versions behind...LOL

Remember to update your libraries and boards as well. That'll keep everything happier...:-)

AlexanderDNye, 2018-08-08:
Ok, so I've gotten a little bit further. 

It seemed that some of the libraries like the ArduinoJson didn't like to be at their latest build. I ended up closely looking at what versions of each library Chris used in his video and used that.

Now its compiling but I'm getting another error.

`Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Archiving built core (caching) in: C:\Users\alexa\AppData\Local\Temp\arduino_cache_476758\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_f61d43f952a7d3551d2738c9a0d3b1b6.a
Sketch uses 471185 bytes (45%) of program storage space. Maximum is 1044464 bytes.
Global variables use 49112 bytes (59%) of dynamic memory, leaving 32808 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

AlexanderDNye, 2018-08-08:
Fixed by resetting the D1 mini.

Thanks for all of everyone's help.

closed 2018-08-08 by AlexanderDNye