Qrome/printer-monitor #44: [Request] Print ETA

SillyHats, 2018-09-10:
It would be great if in addition to Printing Time and Time Remaining the print monitor had an ETA screen with displaying (current time+time remaining).

NeoRame, 2019-07-20:
Any news about that? or any ideas to implement it easily?

Dougsterde, 2021-03-07:
Second this request.

samwiseg0, 2021-06-14:
Bumping as I would like to see an ETA as well!

Dougsterde, 2021-06-15:
Victor7376's fork does this beautifully. Just download and upload it over existing and it will add more screens. I've been using it for a couple of months.

samwiseg0, 2021-06-15:
> Victor7376's fork does this beautifully. Just download and upload it over existing and it will add more screens. I've been using it for a couple of months.

Yep. I noticed that. I have compiled it and its working well. 

Thanks