Qrome/printer-monitor #53: Screen does not work!

Andreiva13, 2018-11-04:
Hello!
The screen just doesn't work tried different pins added pull-up resistors 5v or 3.3v 
But if i use the adrafruit ssd1306 library it works.
Any tips?
TIA

Qrome, 2018-11-05:
The project uses the ssd1306 library.

Andreiva13, 2018-11-05:
I know but that library does not work with my oled

Andreiva13, 2018-11-05:
Might be because my display is blue with yellow instead of white? 
Or that it uses sck instead of scl? I believe that is not case but who knows ?

Qrome, 2018-11-05:
How many pins are on your display?

Andreiva13, 2018-11-05:
It is i2c also to answer your question it has 2 data pins and 2 power pins

Qrome, 2018-11-05:
Yeah, just wanted to make sure.  I have had others with the SPI try to use this.  I have 2 of the blue and yellow displays though my pin is scl -- not sure why that would make a difference.  Is it possible your board has a different address?

The default setting in the Settings.h file is:

const int I2C_DISPLAY_ADDRESS = 0x3c;

Maybe your board has a different display address.

Andreiva13, 2018-11-05:
I tried 0x3d too and if i remember right it has the address 0x3c as i scaned it

Andreiva13, 2018-11-06:
Any idea?

Qrome, 2018-11-06:
Sorry, I have given you my ideas.  Thousands of these have been made.  I personally have made about 20 -- no issues.  Maybe try another screen or another Wemos.

Andreiva13, 2018-11-06:
Yeah thanks i do not have another wemos and I tried with every screen I have thanks for the given time

FischOderAal, 2018-11-12:
For me it does not work as well. If I use the example SSD1306ClockDemo it works fine with the adress 0x3c and pins D2 and D5. But if I upload the printermonitor the screen does not update.

What are the library versions when this is known to work?

Bazeone, 2018-12-04:
have you tried D1 instead of D5? worked on my board.

Andreiva13, 2018-12-04:
Tried that

Niil78, 2018-12-10:
Hi have this problem. i change D5 for D1, and no work... :(

Qrome, 2018-12-15:
Make sure you are using SP8266_and_ESP32_Oled_Driver_for_SSD1306_display version 4.

Andreiva13, 2018-12-15:
Useing that one thanks anyway

Tig3rch3n, 2018-12-16:
same Problem, everything else works by now but not the Display
I tried the i2c scanner to see if my screen has a different address... but no, x 3c is correct...
The bundle I bought contained 10 Diyplays, checked 3 and non of them worked.
With the normal Adafruit and SSD1306 demo i get the demo loop working first try U_u

Qrome, 2018-12-16:
Hmmm, might be cheaper if you can send me a link to the ones you bought.

Qrome, 2018-12-17:
Ok, I ordered one.  Could you post a photo of your wiring and board for me?

jascdk, 2018-12-23:
I have exactly the same problem. Started off using D1 and D2 and got it to work. Then I updated with the new precompiled BIN and the screen went black:( Then I deleted all flash and removed to the D5 pin instead - but this does not work either. I can't figure it out:(

jascdk, 2018-12-27:
> Hello!
> The screen just doesn't work tried different pins added pull-up resistors 5v or 3.3v
> But if i use the adrafruit ssd1306 library it works.
> Any tips?
> TIA

I think I got the trick now:) At least it worked for me the other day! Download the ESP Easy Firmware form github and extract the folder. Then copy all the files for Qrome and put the right compiled Binary (for your screen type) over in the folder you extracted from ESP Easy. 

Then connect the Qrome to the computer and open the ESP8266flash exe file . It should select the right com port, and then you select the compiled binary in the lowest drop down menu.

Then hit flash and it should work.

It looks like I had trouble with getting the right libraries through Arduino IDE:(

This method worked for me - let me hear how it works for you:)

lekrom, 2018-12-27:
Hi All. I will be building this as soon as I get back from summer vacation in a week or so. I have a couple of d1 mini's and an oled or two lying around.
If I may offer my 2c of advice, I know a lot of folks have moved away from arduino ide to platform io, myself included. From what i have read on other projects, the arduino ide can be somewhat shaky with some esp8266 libs. Also, PIO seems to be a bit more efficient in many cases. May be worth looking into. In any case, I plan to port this to platformio. Not too hard to do...

Qrome, 2018-12-27:
@jascdk -- thanks for the update.  I suspected it was coming down to incorrect libraries.  I am going to consider packaging the required libraries with the release.   Thanks for the update.

wesselah, 2019-01-04:
I did some testing because i have the same problem oled 1306 not working. Did the same testing and oled is working with SSD1306ClockDemo it works fine with the adress 0x3c and pins D2 and D5
But i did have an old printermonitor still working version 2.1 i tried the screen on that version and it worked. I could not compile that version anymoe because then i did get this error with the timeclient seems a library problem

wesselah, 2019-01-05:
did use the workarround jascdk  did and that did  work for me so just used the bin file. Had to rename the bin file to get it working with the flash.cmd file from espeasy
so it must be a library issue.
I tried to go backt to the 2.1 version but that did not compile anymore big issues with the timeclient lib.
Tried 2.2 and 2.2a and that compiled but oled not working and wifimager did not start anymore.
But i have to say that there where a lot of arduino updates and that could be the problem and had to go back to an older json lib to get it compile anyway.
so again thanks for youre work but the espeasy way maybe should be the way foreward.

Qrome, 2019-01-06:
Please provide the error messages you get.   That will be a HUGE help.

iansexton, 2019-02-01:
Be aware that SSD1306 displays DO NOT all have the same pin connections!
There are two common variants. Be sure about your wiring.

Wollemania, 2019-02-26:
Here the same...screen is not willing to work. Bought the suggestet parts from amazon. First of all the pins for SDL and SCA are here D1 for SCL and D2 for SDA. Then the display does not work with the library. Here i tested with a sketch from a amazon buyer of the 1.3 display. The display works with the U8x8lib. But how to implement this correctly in the printermonitor sketch? I´m quite new with arduino.

Qrome, 2019-02-26:
Either wire it up like it shows on the readme file or change the code to use D1 and D2 for the I2C connection to the screen.

Qrome, 2019-02-26:
I still have yet to see a standard I2C SSD1306 screen that would not work with this project.   I have made many and tried many different models.

Wollemania, 2019-02-27:
I´m sure about the wiring, tested it with a sketch to find out if screen com. works. I modified your sketch for 1.3 display and changed the PIN definition in configuration. But the web interface, wifi and so on works fine. I think I have to check further on.

iansexton, 2019-03-05:
@Qrome What is a 'standard' I2C SSD1306??? More importantly. What pinout does YOUR standard use?

Qrome, 2019-03-05:
I2C has standard pins -- the code does not use any special drivers other than the libs mentioned.   You also have the ability in the code to change the pins to almost anything you want (except the pin that is used for the internal LED).   There are TONS of people building these every day and posting their builds and working.   I have yet to find a SSD1306 that does not work with it.   Builds are shared daily on Thingiverse.     Use the SSD1306 mentioned in the Readme file.   I have used several from AliExpress as well without any issues.   If you have issues, please post the details including photos.

Qrome, 2019-03-05:
I have seen people try to use the SSD1306 with the SPI interface -- and those do not work.

juanillo62gm, 2019-03-09:
I'm trying with the SSD1306 (https://www.banggood.com/7Pin-0_96-Inch-OLED-Display-12864-SSD1306-SPI-IIC-Serial-LCD-Screen-Module-For-Arduino-p-1364267.html?rmmds=myorder&cur_warehouse=CN)

Using this wiring:
[image]

Screen isn't working ;(

wabbitguy, 2019-03-09:
That SSD1306 you linked to has the jumpers on the back of the display set for 4 wire SPI, not I2C...

I2C displays only have 4 pins on them. Like these ones:

https://www.banggood.com/3Pcs-0_96-Inch-4Pin-IIC-I2C-Blue-OLED-Display-Module-For-Arduino-p-1162507.html

Qrome, 2019-03-09:
Please use the 4 pin I2C as mentioned in the Readme file.   I have not tested nor have I recommended use of the SPI models.  I know they have jumpers on the back -- you may want to check those, but not sure if the drivers are the same for them or not as I have not used or tested them.

Wollemania, 2019-03-09:
Hello together,

finally got my printer monitor working.
After getting sure the wiring is correct I searched for errors in the Arduino configuration.
So this is what I did:
1.) First check the pin coniguration for the ESP8266. My board uses the Pins D1 for SCl, and D2 for SDA. If so correct this in Settings.h

2.) If using 1.3" display uncomment line #define DISPLAY_SH1106 (qrome made a great job in commenting each line)

Now here is my fault or why my monitor does not work.

3.) In board manager be sure to use the ESP8266 board driver version 2.4.1 However the newer versions use the LOLIN (WEMOS) boards which are perhaps not supported. In the 2.4.1 version the right board WeMos D1 R2&mini can be selected. 

Now I´m happy with a full functional printer monitor. 
Thanks to qrome for his patience to reply on all issues. Great job.

Qrome, 2019-03-09:
@Wollemania thanks for sharing your experience.   You should be able to use the ESP8266 core version 2.5.0 as well -- I have tested this just fine.    Thanks.

wabbitguy, 2019-03-11:
The seven pin SPI displays work just fine with printer-monitor but require some changes in the code selections and they need to be wired properly.

Settings.h - additions:

#include <SPI.h>
#include "SSD1306Spi.h"

// Display Settings
const int I2C_DISPLAY_ADDRESS = 0x3c; // I2C Address of your Display (usually 0x3c or 0x3d)
const int SDA_PIN = D2;
const int SCL_PIN = D5;
boolean INVERT_DISPLAY = false; // true = pins at top | false = pins at the bottom
//#define DISPLAY_SH1106       // Uncomment this line to use the SH1106 display -- SSD1306 is used by default and is most common

#define DISPLAY_OLED7Spi // this is for the seven pin SPI OLED displays (combo I2C and SPI)
// WeMOS D5 --> OLED D0
// WeMOS D7 --> OLED D1
#define OLED_DC     5  //WeMOS D1 Connect to DC on OLED
#define OLED_CS     15 //WeMOS D8 Connect to CS on OLED
#define OLED_RESET  0  //WeMOS D3 Connect to RES on OLED

printermonitor.ino changes:

// Initialize the oled display for I2C_DISPLAY_ADDRESS
// SDA_PIN and SCL_PIN
#if defined(DISPLAY_OLED7Spi)// check for SPI display type first, then the I2C types
SSD1306Spi display(OLED_RESET, OLED_DC, OLED_CS);
#else
#if defined(DISPLAY_SH1106)
SH1106Wire     display(I2C_DISPLAY_ADDRESS, SDA_PIN, SCL_PIN);
#else
SSD1306Wire     display(I2C_DISPLAY_ADDRESS, SDA_PIN, SCL_PIN); // this is the default
#endif
#endif
-----
For the RST, DC and CS definitions, I never could get the D# pins to work so I used the standard GPIO pin numbering and it works perfectly. Not sure why, and never bothering looking in the libraries to find out why.

There's also an SPI 6 pin OLED that I haven't tested (they call it a 4 wire SPI interface). And again it's a combo display with resistors/jumper points on the back of the PCB for I2C or SPI.

[image]

iansexton, 2019-03-14:
> I2C has standard pins -- the code does not use any special drivers other than the libs mentioned. You also have the ability in the code to change the pins to almost anything you want (except the pin [...]

Wow! So defensive :( I2C does not have standard pins! It is an electronic standard - not a physical standard. An I2C device requires power, clock, & data. Not pins!

I'm out of here. You clearly can't accept constructive criticism.

Ian

Qrome, 2019-03-14:
> @Qrome What is a 'standard' I2C SSD1306??? More importantly. What pinout does YOUR standard use?

@iansexton  -- How does one respond to this?   Sorry if my reply seems defensive, but YOUR question was asking about "pinout" -- sorry, but support for something like this gets very tired quick when everyone is at different levels in Arduino development that goes along with so many components that differ greatly in quality.

iansexton, 2019-03-15:
@Qrome I built this and it worked immediately, it helps that I know what I'm doing...
I pointed out that I2C SSD1306 displays are generally available with two different pinouts. Was that not constructive? You have readers struggling to get this working :(
I pointed out that OLED displays have a limited life but you don't believe me (do some research)
I applaud your efforts! It's great that you have loads of happy users! :)
Good luck & goodbye
Ian

juanillo62gm, 2019-03-21:
> The seven pin SPI displays work just fine with printer-monitor but require some changes in the code selections and they need to be wired properly.
>
> Settings.h - additions:
>
> #include <SPI.h>
> #include [...]

Thank Youuu <3 

[image]

I will make a diagram and pull it for everyone that have a 7 pin screen.

Here is the code modified for 7 pin screen:
[Printer Monitor 7 Pin Screen.zip](https://github.com/Qrome/printer-monitor/files/2994276/Printer.Monitor.7.Pin.Screen.zip)

wabbitguy, 2019-03-21:
@juanillo62gm you're very welcome!

Mel

webline, 2019-04-08:
> @Wollemania thanks for sharing your experience. You should be able to use the ESP8266 core version 2.5.0 as well -- I have tested this just fine. Thanks.

I tested with 2.5.1 - screen dark, nothing. Changed to 2.4.1 and everything works fine. So if using the current version in the board manager (and no "WeMos D1 R2&mini" option) you may suggest what board we should select then. WeMos D1 R2 war there but does not do the job. So thanks to @Wollemania ;)

wabbitguy, 2019-04-08:
@webline The latest release of the ESP boards is 2.5.0. Where'd you find 2.5.1 or is it a beta pre-release?

Qrome, 2019-04-08:
Make sure you are selecting 1M SPIFFS -- Also, 2.5.0 -- it works.

webline, 2019-04-08:
yes - sorry - 2.5.0b3 is the last version - and no, for me 2.5.0 didn't do it - with 1M SPIFFS selected.

wabbitguy, 2019-04-08:
Might be a defective display? I've gotten a few of those.

webline, 2019-04-08:
it works if I switch the board manager (it worked with an Arduino, with a Teensy board too). So the display is OK.

wabbitguy, 2019-04-08:
Good to know...I'm using 2.4.2 on one system and 2.5.0 on the other. They are different that's for sure. I find for most of my coding, 2.4.2 works more reliably. My older sketches don't like 2.5.0 at all, so I just use 2.4.2 as my main dev system.

uwgeophys, 2019-04-16:
I just wanted to chime in here, i have the same issue and thus far the only fix has been to role back to the 2.4.1 board definitions. Anything higher doesnt work i just get a blank screen, and they all say LOLION instead. Its all wired up fine and works a dream (over I2C) aslong as I dont bump the board definition higher.

This was not a problem until i needed wpa2-eap support and thus far ive only got that going on 2.5, without WifiManager mind you because that doesnt support it either.

You seem to intimate its fixed, or should work, is that in the dev branch? Or have you not been able to replicate this phenomenon yet?

*sorry for the formatting, on a mobile and the webapp (and my typing) are terrible.

Qrome, 2019-04-16:
What pins are you using?    None of us have been able to replicate this.   I have a feeling there is an older OLED driver or something that is tied to the older version that you have locally.   I have been running the 2.5.0 board code since it came out and use it with both the 0.96" and the 1.3" display without any issue.   Not able to replicate this.   I have 6 of these stations that I test with.

uwgeophys, 2019-04-16:
I will have to come back to you tomorrow morning if you want me to check anything very specific (23:07:13 LT currently), it is on my desk at work.

From memory standard pins, D2 and D5. I'll have to get all the library repo versions in the morning, particularly the OLED one I'm guessing.

A side note during my testing when I thought it could have been the screen, I tried searching for the I2C address and couldn't detect it on d2/d5, I had to use d1/d5 and then I could see it at 0x3c (?think thats right from memory). Not sure if relevant but thought it was worth mentioning. as I said before when it's on 2.4.1 works a charm on d2/d5.

Qrome, 2019-04-16:
D2 and D5 is what I am using and the defaults set in the code.   You an change that in the Settings.h file if you want to move it to D1 and D2.   Thanks for taking the time here.

uwgeophys, 2019-04-16:
As I say the it was just a little sketch mockup that got it to show the i2c address up in the serial window under d1/d5, but the screen never responded on d2/d5 to the query for the i2c address.

I see the point in the code for that, I had a go before when I had 2.5.0 set at the default board type, and no luck in any combination d1/d5 or d2/d5 in getting the screen going.

I will reconfirm in the morning the versions and check the i2c address search on both 2.4.1 and 2.5.0 board versions in both d1/d5 and d2/d5 combinations make sure I didn't miss something.

uwgeophys, 2019-04-17:
So I only had a brief chance at work today to look at this - I am afraid the only thing that I can confirm is what you already knew - I managed to get it to work on 2.5.0 as well. I think the issue was I did not select the 1MBS file area the first time; I think it was on default 0. When it didn't work I re-read the instructions and rolled back to the versions quoted and low and behold it worked. I am afraid I did not have the time to roll it back again and replicate the problem during today I will see what I can do over the easter weekend I will try get the time to take it home from work and do it again.

As a side note I was successful in modifying your code though (now the screen is working in 2.5.0) to ignore the WifiManager; and use a hard coded portion to work with the wpa2 enterprise (peap & mschapv2).

Qrome, 2019-04-17:
Glad you got it working.   I am sad to say that 1M SPIFFS being blank has hit a lot of people.

pforrmi, 2019-08-28:
I've got the same problem and changed the settings to:

**const int SDA_PIN = D2;
const int SCL_PIN = D1;**

So instead of D5 for SCL the D1 worked for me!

[1 more reply]