Qrome/printer-monitor #55: ENABLE_OTA = false results in crash loop

vergoh, 2018-11-18:
Using current master branch revision results in a crash loop after wifi settings have been configured if ENABLE_OTA is set to false in Settings.h. No other settings have been changed. Everything works as expected during manual wifi configuring but once the configuration has been given then a crash loop follows and the web interface can't be accessed for further configuration. This appears to happen only if ENABLE_OTA = false. With the default "true" value everything works as expected.

Tested with Arduino 1.8.7, WiFiManager 0.14.0, esp8266 2.4.2, so essentially the latest ones. Not sure if this issue is library or implementation related.

Serial console shows the following output repeating:
```
18:42:06.482 -> load 0x4010f000, len 1384, room 16 
18:42:06.482 -> tail 8
18:42:06.482 -> chksum 0x2d
18:42:06.482 -> csum 0x2d
18:42:06.515 -> vbb28d4a3
18:42:06.515 -> ~ld
18:42:06.647 -> UtcOffset=-7.00
18:42:06.647 -> OctoPrintApiKey=
18:42:06.647 -> OctoPrintHostName=octopi
18:42:06.647 -> OctoPrintServer=
18:42:06.647 -> OctoPrintPort=80
18:42:06.647 -> OctoAuthUser=
18:42:06.647 -> OctoAuthPass=
18:42:06.647 -> minutesBetweenDataRefresh=15
18:42:06.647 -> themeColor=light-green
18:42:06.681 -> www_username=admin
18:42:06.681 -> www_password=password
18:42:06.681 -> DISPLAYCLOCK=1
18:42:06.681 -> IS_24HOUR=0
18:42:06.681 -> DISPLAYWEATHER=1
18:42:06.681 -> WeatherApiKey=
18:42:06.681 -> CityID: 5304391
18:42:06.681 -> IS_METRIC=0
*WM: 
18:42:06.747 -> *WM: AutoConnect
18:42:06.747 -> *WM: Connecting as wifi client...
18:42:06.747 -> *WM: Using last saved values, should be faster
*WM: Connection result: 
18:42:11.649 -> *WM: 3
18:42:11.649 -> *WM: IP Address:
18:42:11.649 -> *WM: 192.168.0.246
Signal Strength (RSSI): 90%
18:42:11.682 -> Server started
18:42:11.682 -> Use this URL : http://192.168.0.246:80/
18:42:19.201 -> Exception (28):
18:42:19.201 -> epc1=0x4021735a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
18:42:19.234 -> 
18:42:19.234 -> ctx: cont 
18:42:19.234 -> sp: 3ffffbf0 end: 3fffffd0 offset: 01a0
18:42:19.234 -> 
18:42:19.234 -> >>>stack>>>
18:42:19.234 -> 3ffffd90:  3ffffe60 3ffffe50 0000000c 00000002  
18:42:19.234 -> 3ffffda0:  3ffeb676 001863be 00000000 33fffe80  
18:42:19.234 -> 3ffffdb0:  3ffffe80 3ffffe70 00000018 4021dd1c  
18:42:19.234 -> 3ffffdc0:  4021dd08 3fffff5c 3fffff68 40220544  
18:42:19.234 -> 3ffffdd0:  3ffffe82 00000010 3ffe866c 3ffffe00  
18:42:19.268 -> 3ffffde0:  40217353 3fff1ee0 00000001 401004e8  
18:42:19.268 -> 3ffffdf0:  7474685f 3ffe0070 3fff1ee0 402172b0  
18:42:19.268 -> 3ffffe00:  00000010 00000010 00000000 f600a8c0  
18:42:19.268 -> 3ffffe10:  00ffffff 0100a8c0 61636f6c 745f006c  
18:42:19.268 -> 3ffffe20:  3f007063 00000000 00000005 00000004  
18:42:19.301 -> 3ffffe30:  00000005 005f0000 3fff1fd9 40201c3d  
18:42:19.301 -> 3ffffe40:  3ffffdf0 00000001 00000017 00000074  
18:42:19.301 -> 3ffffe50:  40104d40 003c7ba5 3fff2150 4021dd1c  
18:42:19.301 -> 3ffffe60:  3fff1e88 3fff1ee0 00000000 4020cdc7  
18:42:19.301 -> 3ffffe70:  3ffe8524 00000005 4021cbc8 3fffefb0  
18:42:19.334 -> 3ffffe80:  40201d42 000001f4 000001f4 40201d37  
18:42:19.334 -> 3ffffe90:  00000075 00000004 000001f4 4020b6b8  
18:42:19.334 -> 3ffffea0:  4021dd08 3fffff5c 3fff1e00 4021dd1c  
18:42:19.334 -> 3ffffeb0:  4021dd08 3fffff5c 3fff1e00 4020dc14  
18:42:19.334 -> 3ffffec0:  00000000 00000000 3ffeb630 00000000  
18:42:19.367 -> 3ffffed0:  3fff5c9c 0000000f 00000000 3fff5cb4  
18:42:19.367 -> 3ffffee0:  0000000f 00000000 00000000 00000000  
18:42:19.367 -> 3ffffef0:  00000000 4021ee90 00000000 4021ee90  
18:42:19.367 -> 3fffff00:  00000000 4021ee90 00000000 4021ee90  
18:42:19.367 -> 3fffff10:  00000000 4021ee90 00000000 4021ee90  
18:42:19.400 -> 3fffff20:  00000000 00000000 ffffffff fe000001  
18:42:19.400 -> 3fffff30:  3ffe8a52 00000000 fe01ef35 4020d270  
18:42:19.400 -> 3fffff40:  00000000 0000000a 3fff5ccc 00000000  
18:42:19.400 -> 3fffff50:  00000000 00000000 4021dd08 00000000  
18:42:19.400 -> 3fffff60:  00000000 00000000 00000000 00000000  
18:42:19.433 -> 3fffff70:  00000000 00000000 00000000 00000000  
18:42:19.433 -> 3fffff80:  3fff5cfc 0000000f 0000000d 4021ee90  
18:42:19.433 -> 3fffff90:  f600a8c0 feefeffe feefeffe feefeffe  
18:42:19.433 -> 3fffffa0:  feefeffe feefeffe feefeffe 3fff20f4  
18:42:19.433 -> 3fffffb0:  3fffdad0 00000000 3fff20ec 4021cc4c  
18:42:19.466 -> 3fffffc0:  feefeffe feefeffe 3ffe866c 40100739  
18:42:19.466 -> <<<stack<<<
18:42:19.466 -> 
18:42:19.466 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

Qrome, 2018-11-26:
I have confirmed that this is a bug.   It looks like if OTA is not being used we can't call the MDNS object for obtaining hostname.   I have updated the 2.2 branch to fix this.

Qrome, 2018-11-26:
This is now fixed in the 2.2 release.   Please pull down the latest and check it out.   Thanks.

closed 2018-11-26 by Qrome