Qrome/printer-monitor #63: invalid operands of types 'double' and 'long int' to binary 'operator%'

Tig3rch3n, 2018-12-14:
Hey Ho Guys,

I just got all parts out of my Letterbox, threw everything together and was about to program my wemo but I can'nt figure out whats the Problem with this error status U_u
At least there is no issue about this yet

Cheers

> Arduino: 1.8.5 (Windows 10), Board: "WeMos D1 R1, 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"
> 
> sketch\TimeClient.cpp: In member function 'long int TimeClient::getCurrentEpochWithUtcOffset()':
> 
> TimeClient.cpp:151: error: invalid operands of types 'double' and 'long int' to binary 'operator%'
> 
>    return round(getCurrentEpoch() + 3600 * myUtcOffset + 86400L) % 86400L;
> 
>                                                                    ^
> 
> exit status 1
> invalid operands of types 'double' and 'long int' to binary 'operator%'

wywywywy, 2018-12-15:
I literally just had the same problem and I converted the double to longint and it compiles.

TimeClient.cpp line 151
`return (long) round(getCurrentEpoch() + 3600 * myUtcOffset + 86400L) % 86400L;`

Qrome, 2018-12-15:
That file hasn't changed since the initial release.  Honestly, I think you are running with an older core.

wywywywy, 2018-12-15:
Maybe it's too new rather than too old.

It's esp8266 2.5.0-beta1

2.4.2 doesn't have this problem

Tig3rch3n, 2018-12-15:
> Maybe it's too new rather than too old.
> 
> It's esp8266 2.5.0-beta1
> 
> 2.4.2 doesn't have this problem

Yea right, thats the problem U_u
@Qrome maybe you should those Version numbers in the readme, I had another problem but i could solve it because someone else had it as well ;)
I think it was the ArduinoJson Lib

closed 2018-12-15 by Tig3rch3n

Qrome, 2018-12-15:
A couple of notes here.

* I am adding the line to cast to the Long to be compatible with the new 2.5.0-beta ESP826 Core library. 
* The readme.md alreayd states to use the 5.13.X ArduinoJason -- their BETA 6.0 version is a major change.
* The ESP9266 OLED SSD1306 I am using is version 4 -- I have never used version 3.