Qrome/printer-monitor #65: SSH1106 no update after webserver message

sorriso93, 2018-12-18:
Hello
I compiled code v2.3 and put it into a wemos d1 mini with SSH1106 display (I use D3 & D4) and it is all ok till display the message to use the web server with ip address and port... From the serial I see the sketch running, downloading time and weather forecast, all seems ok. On the display I don't get any update after the first message, logo, and ip and port address!

Please note that on the same hardware with the same arduino IDE and same libraries I run the weather station demo from thingpulse (same libraries used as you) and all is ok... Can't figure out what is going wrong...

Qrome, 2018-12-19:
Have you updated the core library for ESP8266 -- look down in the bottom right of your Arduino IDE and see if it says (no SPIFFS)  You must compile with at least 1M for SPIFFS or it can't load settings from the file system.

sorriso93, 2018-12-19:
I'm using esp2866 core 2.4.2, adafruit gfx lib 1.3.6 and oled driver for ssd1306 v4. I recompiled with 2M spiff reserved, same behavior.
Below the log, seems all is ok. The display is freezed on the message to connect to web server http://192.168.34.48:80/.
Maybe it is because can't reach the octoprint server? On the other configuration I tested with SSD1306 all is ok even if the octoprint server is off...

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.34.48
Signal Strength (RSSI): 100%
Server started
Use this URL : http://192.168.34.48:80/
no services found - make sure OctoPrint server is turned on
*** Leaving setup()
*WM: freeing allocated params!

Getting Weather Data...
Getting Weather Data
GET /data/2.5/group?id=3171366&units=metric&cnt=1&APPID=732380cdeb76a5641ddf7599441931a6&lang=it HTTP/1.1
Waiting for data
Response Header: HTTP/1.1 200 OK
lat: 45.18
lon: 9.16
dt: 1545217305
city: Pavia
country: IT
temp: 1.55
humidity: 100
condition: Rain
wind: 0.5
weatherId: 500
description: pioggia leggera
icon: 10d

Updating Time...
.
11:04:21
11:4:21
39861
Local time: 10:04 AM
Getting Octoprint Data
GET /api/job HTTP/1.1
Connection to OctoPrint failed: 192.168.0.200:80

Clock Mode is turned on.
Getting Octoprint Data
GET /api/job HTTP/1.1
Connection to OctoPrint failed: 192.168.0.200:80

Qrome, 2018-12-19:
What happens when you pull up the web interface?   It looks like it is running from the Serial Monitor logs you shared.   Can you try loading the sh1106 compiled binary that was shared in the 2.3 release?

sorriso93, 2018-12-19:
Web interface seems ok and loads al configurations from SPIFF correctly. Using PIN D3 and D4 if I use your binary I will not see anything on screen (you use D6 and D8 PINS). Can you compile a custom version using D3 and D4 PIN?

Qrome, 2018-12-19:
Ok, I know your problem now.   The firmware uses LED_BUILTINand that by default is using D4 (Wemos D1 Mini Default)  This is where you get the blue flashing working LED light.   Well, you have told it that your OLED is now using D4 pin and that is the issue... when the firmware goes to flash the LED -- it locks up your OLED display.   I have seen this before.   You can either change your pins or disable the the built in LED on the Wemos by changing the pin to it in the Settings.h file:

const int externalLight = LED_BUILTIN;

Change LED_BUILTIN to D1 -- that will disable the flashing but let it work.   I would say if you want the LED to flash then change your display pins.

closed 2018-12-19 by Qrome

sorriso93, 2018-12-19:
Confirmed, problem solved changing to D1 the led_builtin! Many thanks

nelscougar, 2018-12-19:
good job guys ! :) 

----- Mail original -----