Qrome/printer-monitor #66: Weather translation

sorriso93, 2018-12-19:
I noticed that in the home page (web interface) there is the english term and local term between () for weather condition, although on the display, no matter what language you set in the settings file, you always have the english term for weather condition.
Is it correct? Some weather condition in italian are very long words and maybe you evaluated not feasible to put it in the display...

Qrome, 2018-12-19:
That is correct... the weather service is always returning English for the short value.   This isn't a bug in the code.   It is a short coming of the weather service.

closed 2018-12-19 by Qrome

sorriso93, 2018-12-19:
Thanks. 
Anyway I prefer the description in local language, so I changed in the printermonitor.c, in the drawWeather with the following:

  //display->drawString(0 + x, 24 + y, weatherClient.getCondition(0)); old instruction
  display->drawString(0 + x, 24 + y, weatherClient.getDescription(0).substring(0,12));

thanks

Qrome, 2018-12-20:
There you go... that way it should always fit on the display.   You may want to check the length of the description or it may throw an exception if it is ever under 12 characters long or if the description happens to be empty when calling the substring.