Qrome/printer-monitor #67: Can't input domain or subdomain

ogland, 2019-01-10:
In the input field "OctoPrint Address (do not include http://)" I can't seem to add domains or subdomains instead of IP. This makes me unable to use different monitors for different instances. I also can't choose which printer to monitor, since it defaults to one of them only.

Please update with this functionality.

Qrome, 2019-01-10:
Only one endpoint is supported.   I will look to see if there is an issue with domains or subdomains, but I have not had issues nor has any one else for the last year.   

Also, feel free to contribute via code and PRs.

Qrome, 2019-01-10:
You have made some incorrect assumptions.  The code does not include the http:// -- it uses the WiFiClient object as this:

printClient.connect(myServer, myPort)

If you are attempting to hit HTTPS -- this is not supported.   You should be able to hit anything that would resolve on the network your printer monitor is on.   Make sure the network is reachable from the same network.

closed 2019-04-12 by Qrome