Qrome/printer-monitor #72: Feedback

iansexton, 2019-01-24:
Hi, I like it so far! :) I have a few minor issues which I'll come back with.
Great work!

Qrome, 2019-01-24:
Please search through and see if the "minor issues" are already covered in other posts.

iansexton, 2019-01-24:
I initially had some issues with libraries - this isn't your fault!
I'm not really interested in weather reports. I have other devices which do this. What's that about?
I'm not so interested in the time of day either, I wear a wristwatch.
These displays don't last very long if they are left on permanently :(
Your compile instructions don't make it clear that SPIFFS is required.
Your code has a lot of constant strings - might these be better as SPIFFS files?
It's confusing that the settings don't take effect unless SPIFFS is erased. I guess it's a case of RTFM but it doesn't 'feel' right.
There are two very common variants of the OLED displays & they have DIFFERENT pinouts. Take a close look on ebay. It's easy to get it wrong following your pictures :(

Great work though :)

Ian

Qrome, 2019-01-24:
1. I have been running these displays constantly for 1.5 years -- no issue with the display failing or going dim -- at all.
2. SPIFFS was default on compiling Wemos D1 boards ONLY until recently.  I will update the README.md file.
3.  The readme clearly states that once the setting.h file is loaded changes are made through the web interface.
4.  All displays work.   The recent branch allows the user to specify the orientation of the displays.

closed 2019-01-24 by Qrome

iansexton, 2019-01-24:
You missed my point.
The pins on the display are often different. NOT the same as your instructions.

Qrome, 2019-01-24:
No, I didn't miss your point.   Thanks.