Qrome/printer-monitor #76: Watchdog resets

airdamien, 2019-02-21:
https://www.amazon.com/gp/product/B01N3P763C/ref=ppx_yo_dt_b_asin_title_o00__o00_s00?ie=UTF8&psc=1

I tried defeating the wdt, among other things, no luck. 

The board never brings up wifi, just loops this.

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

Settings File does not yet exists.
File open failed!
Settings File does not yet exists.
File open failed!
Settings File does not yet exists.
File open failed!
Settings File does not yet exists.
File open failed!

airdamien, 2019-02-21:
hmm, force flashing the 2.3 bin to it seems to have done the trick.

Qrome, 2019-02-21:
You need to set SPIFFS to 1M

Qrome, 2019-02-21:
This is what is says in the readme file under complying the code.

Set 1M SPIFFS -- this project uses SPIFFS for saving and reading configuration settings.
Loading Supporting Library Files in

closed 2019-02-21 by Qrome