Qrome/printer-monitor #78: Error uploading

Dutchy-79, 2019-03-10:
Hi,

I'm trying to get this to work and bought a Wemos and LCD, I eventually changed the adres because at the back it says 0x78

First few attempts didn't gave an error though now it says this.

> Arduino: 1.8.6 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"
> 
> Sketch uses 482068 bytes (46%) of program storage space. Maximum is 1044464 bytes.
> Global variables use 47972 bytes (58%) of dynamic memory, leaving 33948 bytes for local variables. Maximum is 81920 bytes.
> C:\Users\####\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd nodemcu -cb 921600 -cp COM3 -ca 0x00000 -cf C:\Users\####\AppData\Local\Temp\arduino_build_465884/printermonitor.ino.bin 
> esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
> 	setting board to nodemcu
> 	setting baudrate from 115200 to 921600
> 	setting port from COM1 to COM3
> 	setting address from 0x00000000 to 0x00000000
> 	espcomm_upload_file
> 	espcomm_upload_mem
> 	setting serial port timeouts to 1000 ms
> opening bootloader
> resetting board
> An error occurred while uploading the sketch
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> resetting board
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> resetting board
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> trying to connect
> 	flush start
> 	setting serial port timeouts to 1 ms
> 	setting serial port timeouts to 1000 ms
> 	flush complete
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	read 0, requested 1
> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

Device manager says USB SERIAL CH340 COM3 so that should be okay.

Any idea?

Dutchy-79, 2019-03-10:
To make sure I didn't create the error I downloaded a fresh copy and tried once more.
Done Uploading though no image and no Wi-Fi network available.

> Sketch uses 481968 bytes (46%) of program storage space. Maximum is 1044464 bytes.
> Global variables use 47968 bytes (58%) of dynamic memory, leaving 33952 bytes for local variables. Maximum is 81920 bytes.
> Uploading 486112 bytes from C:\Users\####\AppData\Local\Temp\arduino_build_476821/printermonitor.ino.bin to flash at 0x00000000
> ................................................................................ [ 16% ]
> ................................................................................ [ 33% ]
> ................................................................................ [ 50% ]
> ................................................................................ [ 67% ]
> ................................................................................ [ 84% ]
> ...........................................................................      [ 100% ]

Dutchy-79, 2019-03-10:
Okay I kept trying and now I have Wi-Fi setup and still no image.
I have 3 different displays 2 say 0x78 on the back and 1 says nothing. I tried keeping it stock and also changed it to 0x78 and no image with them.

Dutchy-79, 2019-03-10:
To get it to work I had to install a previous version of the board
ESP8266 version 2.4.1 works for the board.

Dutchy-79, 2019-03-10:
one more step, just soldered SCL to D1 and it works.
last issue is the weather saying N/A

Dutchy-79, 2019-03-10:
After 2 hours struggling everything works. Please add you need ESP8266 2.4.1 and pin D1 for SCL if there is no image. weather now works too.

closed 2019-03-10 by Dutchy-79

Qrome, 2019-03-10:
I will update the readme to I dicate you must use ESP8266 core 2.4.1 or higher.  I am using 2.5.0 just fine.  Most everything else is covered in the raedme file.

Dutchy-79, 2019-03-10:
Probably because I used one of the cheapest clones from AliExpress 🤣

Just mentioned it because I didn't want people to say you did a bad job while it's caused by the library and board versions.