Qrome/printer-monitor #80: Some idea's

mikeo999, 2019-03-22:
Create this yesterday and loving it. Thanks for this nice project.

Some idea's
1. Would it be possible to create an option to turn of the screen at specific times. One box is in my bedroom and it would be nice to set a time period so the screen goes dark with sleeping times.

2. Are there option to refresh the screen different way. One box stand by my computer so i can keep an eye on the printing process. Because the info scroll all the time it distract.  Would be nice if it is possible to choose an effect.

mikeo999, 2019-03-28:
3. End Time when printing is finished. Time Reaming is of course also great but sometime it is nice to see when you can start the next print if you a in a hurry.

madzur, 2019-11-27:
Another idea to your point No.1 would be an option to decrease display brightness at night. I am using this on my nightstand and the default display brightness lights up the room.
Would be great if there was an option in the web interface to enable display dimming at certain time intervals. Let's say 50% brightness between 10PM and 6 AM.

HeyRobb, 2020-04-06:
> 1. End Time when printing is finished. Time Reaming is of course also great but sometime it is nice to see when you can start the next print if you a in a hurry.

I too would love to see End-Time option in place of hours/minutes! Thank you!!