Qrome/printer-monitor #82: Repetier Server

kg4iae, 2019-04-05:
I was wondering if anyone has made this work with Repetier Server?  I know it has an API similar to Octoprint.  

I move to Octoprint last year after a Prusa firmware killed Repetier Server.  Supposedly they have the issue fixed, and I was considering moving back, since I do have a paid license.  

I built one of these Printer Monitors, and I really like it.

Qrome, 2019-04-05:
I have a buddy that claimed he was going to do the Repeitier client class -- but never came about.   I hope someone that uses that service will put in a PR.

kg4iae, 2019-04-06:
I was hoping someone else was already working on this.  I'm not very proficient in C++ but I can look into it.  According to the docs, it looks like all I need to do is a json web request to the server, with an API key and the printer name.  (Repetier lets you attach multiple printers to a single server).  I can try and write some modules, but I may need help integrating it.

kg4iae, 2019-04-07:
I have my Repetier Server running.  I have found the appropriate API calls to get temps, time elapsed, time remaining, and percentage.  I'm going to work on building a module soon as I get some time.

kg4iae, 2019-04-10:
I have the module mostly done.  It works exactly same way your Octoprint Module.  Repetier supports multiple printers.  I specified in the code to use the first one.  I suppose an enhancement could be made to support multiple printers, but I think that will take some work.

wabbitguy, 2019-04-10:
Need a tester?

kg4iae, 2019-04-10:
I can't seem to figure out how to use Git.  I'm trying to upload the code, but not getting anywhere.

Qrome, 2019-04-10:
@kg4iae -- thanks for being willing to contribute.   I see you made a Pull Request (PR) for my dev 2.6 branch to be merged into master -- this will be done, but not just yet.   What I would like you to do is.

1. Fork from the 2.6 dev release branch that I have -- this will create a branch under your ID in GitHub.
2. Apply your changes and new class file to your own branch.
3. Then create a Pull Request on the 2.6 release branch.

From there we can review and test the changes before we merge it and commit it to the master branch.
Once we get it into the 2.6 -- I think we have some people that can test it as well as polish up the web UI -- to handle selecting what monitor to use.

Thanks.

kg4iae, 2019-04-10:
Thanks for the instructions.  I believe I have done it right this time.

Qrome, 2019-04-11:
I have included the Repetier Client code in a new 3.0 release branch.   I have decided to bump the major version as this will reset the configuration settings for the printer information.   I renamed all the variables to make them generic to the type of server.   

At this point, you can compile for either OctoPrint (default) or uncomment the following line in the Settings.h file to use the Repetier client code.

```//#define USE_REPETIER_CLIENT       // Uncomment this line to use the Repetier Printer Server -- OctoPrint is used by default and is most common```

Potential release branch is here:
https://github.com/Qrome/printer-monitor/tree/3.0

Please help me test this out.  I do not currently have a Repetier print server.

kg4iae, 2019-04-13:
I have a pro license, and an extra pi machine sitting around.  I can set up a server with a virtual printer, if you want to play with it.

Qrome, 2019-04-13:
I installed one of my servers with it yesterday -- It gives me the 14 day pro trial.   Quite nice.   
There are a few more things that I would like to do... like allow the user to select one printer from the collection to monitor.  

As of now can you pull down the latest from 3.0 and test it on your end?   Look over the Web UI stuff as well.

Thanks for the offer on the virtual printer -- I am going to have to decide if I need to buy a license or what.

wabbitguy, 2019-04-13:
Works good once you figure out the nuances. Like if you have more than one printer...:-)

[image]

kg4iae, 2019-04-14:
Qrome, in case you don't already know, you can create a virtual printer.  That will allow you to test the interface without printing anything.  All you do is create a printer, when you get to the part (I think it's page 2) where you select the port, You select "Virtual Cartesian" or "Virtual Delta" for the port.  Continue the wizard and you should have a virtual printer.

Wabbitguy,  There is a single place on the Repetier.cpp module where you can select the printer (if you have more than one)  I did not design it to search for the name.  You will have to identify it by number (0 for first printer, 1 for second printer).  It could probably be modified to allow for the user to select the printer in the config.  

I actually do have 2 printers, but I only use one of the at the moment, the other is hopelessly dead.  Needs a new RAMPS board.  I'm just not very motivated to fix it, since it didn't really work all that well to begin with.

Qrome, 2019-04-14:
Thanks for the update -- I actually connected both my printers to it for some testing.   I will add a third though for fun with the virtual printer.

I have made some big changes in the 3.0 branch -- please pull it down and test it if you can.

The Web Configuration page now allows you to test connection and select the printer you want to monitor.   I have made several other minor changes.   Please test it out.

I plan on adding a "Test Connection" for OctoPrint as well.

wabbitguy, 2019-04-14:
@kg4iae Thanks for the info update, I found that array this afternoon when I was getting it to function. The printer I normally use is "1" so a simple change and it found my workhorse...:)

@Qrome I'll haul down the update, stick in my OLED SPI code so I can use it...LOL...

Mel
---

wabbitguy, 2019-04-14:
@Qrome - NICE WORK...Me Like!....the test button is a great add on and love the printer selection!...

Working perfectly here.

Qrome, 2019-04-14:
LOL -- yeah, no kidding.   Glad it is working.   I added a test button for OctoPrint -- but since it does not like the Cross Origin (if you use Basic Auth) I did it as a popup window with the JSON response -- at least for now.   Thanks for all the help.

kg4iae, 2019-04-15:
I just installed your new version on my device.  I really like the printer selection.  So far it's working for me.  

@wabbitguy I noticed the extra precision on the temperatures.  No idea why Repetier thought that was important.  I would think an integer would be enough, or single if you wanted to get nerdy with it.  But double precision seems overkill.   I wonder if it may be a software limitation.   I think it was written in Python.  I have little knowledge of that language.

Qrome, 2019-04-15:
Sounds good.   I updated it a few hours ago... I am thinking it is very close to release.      You should now only see it go in Monitor Mode when the selected printer is printing (both OctoPrint and Repetier) and now when the printer is on you should see "online" on the display and "offline" when it is off.  There was some other minor changes as well.

wabbitguy, 2019-04-15:
@Qrome I like the little short cut of clicking on the gear icon by the server name to jump right to the configure options...the temps could be int's.

@kg4iae I've never done any Python either. But I'm going to use that double precision as bragging rights to my fellow printing friends....LOL

Qrome, 2019-04-15:
Yeah, the latest code I have it trim down the temps to 2 decimal for the web interface.   No big deal.

I started another project tonight.   ;)  Pi-hole Monitor -- not as fun as the printer monitor -- but it is interesting to see the blocked traffic and number of clients on my network.

wabbitguy, 2019-04-15:
Well there goes my bragging rights for temps...:-)

Looking forward to the PiHole monitor (been running a PiHole for a long time now)...

I'm starting to think that with all these monitoring one ups, it might be time to consider strictly a monitoring station with a larger display. A larger TFT split into say 4 zones where any zone can be any item you want to monitor...ouch, I think I just gave myself a code headache...LOL

kordris, 2019-04-20:
I know this might be a stretch but I switched my board to duet wifi. Could this port work on my monitor also?? I loved using my little print monitor.

Qrome, 2019-04-20:
@kordris -- I just did a quick look around for information on the Duet Wifi printing board.   I could not see any REST API documentation -- not saying there isn't any, but I would have to say... no, I don't think the monitor would know how to get the status from the Duet.

wabbitguy, 2019-04-20:
The only thing I could find was this and it's pretty limited.

https://github.com/dc42/RepRapFirmware/blob/dev/src/Duet/Webserver.cpp

There is no REST API that I could find either...

dogson3006, 2020-05-16:
hi there,
is there a .hex file for the 3.0 version? i`ll ask, because at some reason i cant flash my d1 mini with arduino. get allways compiling errors after updating some liberies. thank you

Qrome, 2020-05-16:
@dogson3006 -- yeah, sorry, 3.0 is not an official release just yet.  Are you planning on using it for connecting to a Repetier Server or OctoPrint?   When you compile it, you have to decide.   Also, if you set your ESP8266 Core back to 2.5.2 version you probably will be able to compile it.

dogson3006, 2020-05-17:
i want to use it on a repetier server. i activated it by deleting the // in front of a row in the setup.h. 
will try to test the other core to compile it, thank you.

--edit--
i downgraded the esp core  to 2.5.2 and i get the same error ESP8266wifi.h: No such file or directory.
then i downgraded the ide to 1.85 but it doesnt work too. 
---edit end---