Qrome/printer-monitor #85: Display weather and clock

marekmeli, 2019-04-29:
Is is possible to display both weather and clock (side by side or in cycles) by enabling them at the same time? I have no way of testing now.
Thank you for your amazing job and excellent user support.

Qrome, 2019-04-29:
Yes, if you build the double wide, it will display the Clock and Temp at the same time.

https://www.thingiverse.com/thing:2989552

[image]

Qrome, 2019-04-29:
Not sure what you mean, it switches between clock and weather every 10 seconds or so when it isn't printing.

marekmeli, 2019-04-29:
Thank you, that's exactly the information I needed.

closed 2019-04-29 by marekmeli