Qrome/printer-monitor #86: Default values

Uniseco, 2019-05-03:
Hello,
I was change the user Admin and Password, but now I don't remimber. How I can restart from zero all the proces?

Qrome, 2019-05-03:
You can plug it into the computer and run the Arduino Port Monitor and when it boots up it will spit out the user and password.

Uniseco, 2019-05-03:
Thank you for your support but I completly noob. How I Can do this?

Qrome, 2019-05-03:
Connect the Printer Monitor up to your USB port just like you did when you loaded it.   Open the Arduino IDE and then under the Tools -> Ports  --- select the com port that the device is connected to.   Then under Tools -> Serial Monitor will pull up a display for the serial monitor.   Make sure the baud rate is set to 115200 for communication (that is at the bottom of the Serial Monitor window).

From there, hit the reset button on the Wemos D1 Mini and it will reboot and as it is booting, it will display the settings in the Serial Monitor window.   Good luck.

closed 2019-05-03 by Qrome

rogodra, 2020-03-10:
Hello, I have found the password as you have indicated. But it doesn't work, ask the password again. How can I reset the password?

Thank you!

Qrome, 2020-03-10:
Make sure you don't have your browser defaulting the password in there.   You can reset all the settings by writing over the config.txt file.   Or do a clean flash.

rogodra, 2020-03-10:
Can you tell me how to do some of these steps?

Thanks!