Qrome/printer-monitor #88: Version of libraries

Banzaizaizai, 2019-05-11:
Hello
please tell what version of needfull scetch i must install. Monitor star, but time not sync, and data from octoprint refreshed with freez

sorry fo bad English

danielmaxsteel, 2019-06-25:
When I'm understanding you correctly i have the same issue: 
I am configuring the wifi connection by connecting to the AP and saving the Wifi Password. Afterwards the D1 mini restarts but the Webserver isn't available. I can see in the terminal that he gets the correct IP Address from the router. When I'm rebooting the ESP8266 the following messages are written in the Serial monitor (note the time difference after "Using last saved values". The IP Address now is not correct. I think the connection to the router somehow doesn't work. I already tried a different D1 mini as well as different versions of the 8266 libs and wifimanager. 

18:29:20.807 -> UtcOffset=2.00
18:29:20.807 -> OctoPrintApiKey=
18:29:20.807 -> OctoPrintHostName=octopi
18:29:20.807 -> OctoPrintServer=
18:29:20.807 -> OctoPrintPort=80
18:29:20.807 -> OctoAuthUser=
18:29:20.807 -> OctoAuthPass=
18:29:20.807 -> minutesBetweenDataRefresh=15
18:29:20.807 -> themeColor=light-green
18:29:20.841 -> IS_BASIC_AUTH=1
18:29:20.841 -> www_username=admin
18:29:20.841 -> www_password=Password
18:29:20.841 -> DISPLAYCLOCK=1
18:29:20.841 -> IS_24HOUR=0
18:29:20.841 -> INVERT_DISPLAY=0
18:29:20.841 -> DISPLAYWEATHER=1
18:29:20.841 -> WeatherApiKey=
18:29:20.841 -> CityID: 5304391
18:29:20.841 -> IS_METRIC=1
18:29:20.841 -> WeatherLanguage=de
18:29:20.841 -> HAS_PSU=0
18:29:20.908 -> *WM: 
18:29:20.908 -> *WM: AutoConnect
18:29:20.908 -> *WM: Connecting as wifi client...
18:29:20.908 -> *WM: Using last saved values, should be faster
18:33:30.631 -> *WM: Connection result: 
18:33:30.631 -> *WM: 3
18:33:30.631 -> *WM: IP Address:
18:33:30.631 -> *WM: 169.254.234.241
18:33:30.699 -> Signal Strength (RSSI): 88%
18:33:30.699 -> Server started
18:33:30.699 -> Use this URL : http://169.254.234.241:80/
18:33:39.215 -> *** Looking for octopi over mDNS
18:33:39.215 -> Found: octopi.local
18:33:39.215 -> Found: am335x-opt.local
18:33:39.215 -> *** Leaving setup()
18:33:39.215 -> *WM: freeing allocated params!
18:33:39.249 -> 
18:33:39.249 -> Getting Weather Data...
18:33:39.249 -> Getting Weather Data
18:33:39.249 -> GET /data/2.5/group?id=5304391&units=metric&cnt=1&APPID=&lang=de HTTP/1.1
18:33:39.249 -> connection for weather data failed
18:33:39.249 -> 
18:33:39.249 -> Updating Time...
18:33:39.249 -> connection failed
18:33:39.249 -> Local time: 12:-- AM
18:33:39.249 -> Clock Mode is turned on.

Qrome, 2019-06-25:
@danielmaxsteel -- I don't think your issue is related at all to what was mentioned before.

The IP address your Printer Monitor has is NOT a real IP.  Note it is 169.254.234.241 -- is this a valid IP on your network?

Please read this:  https://www.webopedia.com/TERM/A/APIPA.html

You need to make sure your device obtains a valid IP.

Qrome, 2019-06-25:
This might help you out as well:  https://appuals.com/how-to-fix-ip-address-starting-with-169-254/

danielmaxsteel, 2019-06-25:
I'm sorry, you are right. I connected my D1 mini with a hotspot from my mobile phone and were successfully able to access the webserver. So now I need to find the reason why my FritzBox gives the D1 mini no IP Address... But that's no problem with your code!