Qrome/printer-monitor #89: Progress bar does not match Octoprint

Jeff-Haas, 2019-06-25:
Progress bar on Printer Monitor does not match the one shown in Octoprint.

Note I have PrintTimeGenius Plugin installed on Octoprint.

[image]

Qrome, 2019-06-25:
Does print genius plugin change the percentage?   The monitor displays the value from progress->completion value in the /api/job call.   It matches my display value correctly.

I have a feeling your plugin has altered the values to a different completion estimate.

Jeff-Haas, 2019-06-25:
I think you're right, there's something different about the way PrintTimeGenius reports the progress bar.  Note that you are showing the same time remaining as Octoprint, which is using PrintTimeGenius, it's just how the progress bar is being reported.

Here is his explanation of his process:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/How-does-it-work%3F

What I'm seeing is that PrintTimeGenius does a much better job of reporting estimated completion times than the standard estimate.

I'm going to drop him a note and see if we can figure out what this is.

Jeff-Haas, 2019-06-27:
I think I figured it out - you must be using the standard Octoprint estimate, instead of having access to the one from Print Time Genius.  I watched a short print of about 23 minutes, and the regular Octoprint estimate was lagging behind the Print Time Genius estimate.  But near the end of the print, the Octoprint estimate catches up to the actual amount of time remaining.  This seems to be typical behavior for Octoprint.

eyal0, 2019-06-27:
@Qrome Yes, PTG does it in javascript: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/137#issuecomment-506434934

https://github.com/eyal0/OctoPrint-PrintTimeGenius/commit/cd41885348e0a27005c73ba67d2af4fdce28dbdd

Qrome, 2019-06-27:
Thanks for the details.

At this point there is no plan to try and read the progress data from the PrintTimeGenious plugin.   Thanks.

closed 2019-06-27 by Qrome

eyal0, 2019-06-28:
Makes sense.  What you could do, if you wanted, is to have the device divide the time spent from the total time reported by octoprint and report that instead of the percentage.

Qrome, 2019-06-28:
Yeah, the problem is... the total time estimate is actually very inaccurate and changes dramatically especially at the beginning of the print.   Thanks for the suggestions.

eyal0, 2019-06-29:
Yes, I know that the default print time estimate is very inaccurate.  That's why people use Print Time Genius to get better estimates.  The the [graph](https://plugins.octoprint.org/plugins/PrintTimeGenius/) at the bottom you can see how the default estimate moves around while the PTG estimate is very close to the actual time to print.

Maybe it could be an option to use print time instead of file size?

Qrome, 2019-06-29:
Thanks, it might be possible to have it as an option, but currently the Printer Monitor only reads the data from the JSON response from the octoprint API -- does the plugin alter the data that is returned from the standard api?

eyal0, 2019-07-03:
Yes it does.  The estimate total print time and the time to print come from OctoPrint and my plugin (PrintTimeGenius) is making a [print time estimator](https://docs.octoprint.org/en/master/plugins/hooks.html?highlight=PrintTimeEstimator#octoprint-printer-estimation-factory) that will replace the estimator that is in OctoPrint.  The JSON response from OctoPrint will use the numbers that PTG has provided.

You can see in the image that the print time estimation on the device matches the one in the screen.