Qrome/printer-monitor #91: Compile problems

NeoRame, 2019-07-18:
I tried to compile the latest version. Everything seams to be ok, I got no error messages, the upload is fine and the serial monitor give me the right output BUT the display doesn’t work. 
When I reupload the precompiled .bin from your git, everything work fine. Any idea why this is happens? 

Kind regards
NeoRame

Qrome, 2019-07-18:
What version of the OLED display are you using?   What version is your ESP8266 Core Library?   I would check the versions if your libs.

Qrome, 2019-07-18:
Also, do you have 1M SPIFF set -- if you do not have this set when you compile it could cause you problems like this.

NeoRame, 2019-07-18:
THX for the fast response..... 

I got -.-'

My D1 isn't a Original. I changed the Board to "LOLIN(WEMOS) D1 R2 & mini" since then it works fine.

closed 2019-07-18 by NeoRame