Qrome/printer-monitor #96: Status LED/Neopixel, flash when done

PunyViking, 2019-10-07:
I love this project! Got a all in one TTGO, esp8266 with a oled and battery charger/booster!

I want to build a few of these (9 pcs) for all the printers in our makerspace, so that the users can bring a printer-monitor with them anywhere inside the makerspace.

But I wondered if it would be possible to make a external LED/Neopixel flash when the print is done?

I good with hardware, but not with software so I gave up trying to implement this myself

Qrome, 2019-10-07:
I would imagine you would want to have the LED either on of off with active printing.   It would not be hard to associate it with the active print.   Sounds like fun.   I don't have time to implement something like this right now, but it would not be difficult.

shadowCZ75, 2020-03-12:
> Ich würde mir vorstellen, dass Sie die LED beim aktiven Drucken entweder ein- oder ausschalten möchten. Es wäre nicht schwer, es mit dem aktiven Druck zu verknüpfen. Hört sich nach Spaß an. Ich habe momentan keine Zeit, so etwas umzusetzen, aber es wäre nicht schwierig.

nice ;)

highground88, 2020-03-16:
@shadowCZ75 Take a look at this project - you should be able to pull the necessary code to do this (and leave the Prwol stuff out). If you do so, post it up for everyone perhaps :-)

highground88, 2020-04-25:
Here you go @PunyViking , I made a version in which an external LED will flash upon completion.  I hacked together code and unfortunately can't find the other donor... sorry to whoever it was for lack of credit.

@Qrome  - If you're ever making changes to your code I find this a worthwhile feature :-) Great job.

micheld93, 2021-06-06:
Hi everybody, 

I tried to implement all things from the link @highground88 mentioned. Unfortunately, I get no Prowl-notification and the LED is not blinking. I checked the code from the PDF twice and was not able to see any errors. The code is compiled without error messages. The printing job status is displayed correctly. Is there someone with working files with this modification?