Qrome/printer-monitor #97: after 2.5 upgrade cannot get weather info

papperone, 2019-10-11:
I upgraded from old 2.1 version to 2.5 my setup and since then I can't see any info about weather; no icons, 0°C and N/A is only I've displayed on OLED and in the main webpage of printmonitor I've only "Please Configure Weather API".
I'm sure I setup correctly API and city info (as they were in the previous version) and I tested that they do work from my browser with this URL "http://api.openweathermap.org/data/2.5/forecast?id=<CITYCODE>&APPID=<MYAPIKEY>"
and I get the JSON with all the right data.
Can you help to fix this behaviour?

shadowCZ75, 2020-03-12:
I have the same problem again.
--

SohaibSiddique, 2022-01-17:
I have also faced the same problem but while debugging I have added some print lines but with those lines, the weather is working 😄 , it is the problem with the WiFi clients I guess the client would not connect.
FYI @papperone @shadowCZ75 

```
Serial.println("Getting Weather Data");
  Serial.println(apiGetData);
  Serial.println("server name below");
  Serial.println(servername);
  Serial.println(weatherClient.connect(servername, 80));
```