Qrome/printer-monitor #98: Managing multiple Settings.h files

jpasqua, 2019-10-25:
Love the printermonitor and I have forked a copy to make some changes / additions. For example, I've modified OpenWeatherClient to extract the timezone from the weather response. This allows the timezone to be set automatically (assuming your printer is in the same place as the weather you've chosen to monitor). I added an option for this in the Web UI.

Anyway, I am using printermonitor on several different devices with different pins and other minor changes. If I want to submit a pull request, I'll want my Settings.h file to be as vanilla as possible and not include things like my OWM api key. Many settings can be left as default and set through the UI, but not all. Do you have a suggested method for dealing with multiple versions of this file?

Qrome, 2019-10-26:
I too had once used the weather service to get the timezone -- however not all the international time zones from the weather service were correct and had many international users complaining.

Thanks for the offer.   My time is limited for this project right now.   I have not had any request that I know of for "profiles" around different settings files.   Thanks though.

jpasqua, 2019-10-27:
Thanks for your great work on this project. I took a fork and have made my changes there.