Qrome/printer-monitor #101: Webserver not working when compiling from source

Neukam, 2019-12-02:
When I compile from source using Arduino 1.8.10 the web server will not load.  The wifi manager works, and it will connect to my network.  If I put my settings into Settings.h, it connects to my printer and displays properly, but the web page will not load.  If I flash the pre-compiled binary, the web server works fine.  I suspect the issue is cause by using different library versions.

Qrome, 2019-12-02:
Have you tried compiling it with the library versions mentioned in the Readme.md file?

Neukam, 2019-12-03:
The only library version mentioned in the Readme.md file is ArduinoJson (version 5.13.1) which was included.  The other libraries are only mentioned by name and do not specify a version number

Qrome, 2019-12-03:
Sorry, it is mentioned in one of my other projects.  It was compiled with ESP8266 core 2.5.2 -- but I don't think that is the problem.   Do you know what ESP8266 core version you have loaded? 

When you compile it yourself, are you turning on SPIFFS 1M?

Neukam, 2019-12-03:
I was using ESP8266 2.6.2.  Changing to 2.5.2 solved the problem.  Thanks for the help, and thank you for the helpful project.

closed 2019-12-03 by Neukam