Qrome/printer-monitor #109: Add more variable queries

G-Ray2001, 2020-07-01:
Hello,

This project is awesome! It took me awhile to determine why my 1306 wasn't displaying anything, then found out that I needed to use the defaults for SCL and SDA, then had to modify the SSD1306wire declaration line.

I have a question though: I'm running the Octoprint Enclosure plugin as well and have a two temperature sensors, inside and outside the enclosure I'm building. I may also change these to temp-humidity sensors. Is there a way to add these variables into the 1306 display? I'm not familiar with Arduinojson, pretty familiar with Arduino programming though. If this is doable, please offer suggestions on how I can make this mod.

Thank you,

Gamma

Qrome, 2020-07-01:
When running plugins, you're going to need to find out if the values you are interested are exposed through an API call with the rest service JSON response. If that is exposed and you can get those values you could modify it to display it. Sorry but I do not have any plans to implement this.

G-Ray2001, 2020-07-01:
Hello,

Thank you for your quick response. API calls can be utilized with the Enclosure plugin, but It appears to be directed at output actions only, used for printer control via R-PI GPIO pins, not data. I'll investigate further and contact the author of the Enclousre plugin.

Is there particular section of your code I need to focus on if I'm able to get the values through an API call?

Thank you again,

gamma