Qrome/printer-monitor #110: displays only Web interface on ip

Hairybiker, 2020-07-03:
New install, setup & connected to octiopi fine.
Web page reads status fine, but the oled only displays 
Web Interface On
You may connect to IP 
192.168.1.209
Port 80

No status or weather.

Web page displays
Time: 14:34:07
Host Name: octopi
Status: Offline

Galashiels, GB
light rain
1% Humidity
1.34 kph Wind
Rain (light rain)
14°C

How do I get the oled to update?

Qrome, 2020-07-03:
You might be having a hardware issue. If you connect and use the port monitor and you see that it is actively getting the weather and displaying all that stuff, then the wemos is running. It might be the display.

Make sure that you also compiled it with spiff 1M enabled.

Hairybiker, 2020-07-03:
It getting the weather data etc
"country: GB
temp: 14.61
humidity: 1
condition: Rain
wind: 6.49
weatherId: 500
description: light rain
icon: 10d

Updating Time...
.
17:05:11
17:5:11
61511
Local time: 6:05 PM
Signal Strength (RSSI): 100%
Signal Strength (RSSI): 100%
Getting Octoprint Data via GET
GET /api/job HTTP/1.1
Printer Not Operational
Getting Octoprint Data via GET
GET /api/printer?exclude=sd,history HTTP/1.1
Signal Strength (RSSI): 100%
Signal Strength (RSSI): 100%
Getting Octoprint Data via GET
GET /api/job HTTP/1.1
Printer Not Operational
Getting Octoprint Data via GET
GET /api/printer?exclude=sd,history HTTP/1.1
Signal Strength (RSSI): 100%
"
The display was working as a weather station before so I know it works fine.
I have the flash size set to 4MB(FS:1M OTA:~1019KB)
which I assume is the spiff setting as I can't see any other setting in the menu to change it.

The serial monitor is updating every few seconds but no changes on the oled
I have tried with clock on and off and no difference to the display.

Qrome, 2020-07-03:
What pins are you using for the I2C on the display.  If you are using D4 -- that is going to lock the display when the blue LED goes to flash.   The wemos uses pin D4 for the internal LED.

Hairybiker, 2020-07-03:
OK that might be it. I have it on sda/5 and scl/4 which is hardwired at the moment

Qrome, 2020-07-03:
Yeah, that would be it.  This project flashes the internal blue LED as a status indicator.   The moment it goes to flash the LED, it will freeze the OLED if you are trying to use D4.

Qrome, 2020-07-03:
By the way the Wemos D1 mini does not use D4 and D5 for i2c -- the default ones are D1 and D2 as can be seen in the pinout map.
https://escapequotes.net/wp-content/uploads/2016/02/esp8266-wemos-d1-mini-pinout.png

This project uses by default D1 and D5 -- but you can use anything you want other than D4.

Hairybiker, 2020-07-03:
D1 causes a hard reset every few sec.
Trying D2
OK NO display at all now.

Qrome, 2020-07-03:
Use D2 and D5 -- you should be fine.  Sorry, this project defaults are D2 and D5.   That will work for you.

Hairybiker, 2020-07-03:
OK That fixed it thanks for the help

closed 2020-07-03 by Hairybiker