Qrome/printer-monitor #116: Issue compiling

nogardvfx, 2020-08-06:
Hi there, running into an issue and do not know where to really begin to debug it.

In file included from sketch\Settings.h:53:0,
                 from C:\Users\nogardvfx\Downloads\printer-monitor-master\printer-monitor-master\printermonitor\printermonitor.ino:31:
C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SH1106Wire.h: In constructor 'SH1106Wire::SH1106Wire(uint8_t, uint8_t, uint8_t, OLEDDISPLAY_GEOMETRY, HW_I2C, int)':
C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SH1106Wire.h:74:47: error: 'Wire1' was not declared in this scope
       this->_wire = (_i2cBus==I2C_ONE)?&Wire:&Wire1;
                                               ^
In file included from sketch\Settings.h:54:0,
                 from C:\Users\nogardvfx\Downloads\printer-monitor-master\printer-monitor-master\printermonitor\printermonitor.ino:31:
C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SSD1306Wire.h: In constructor 'SSD1306Wire::SSD1306Wire(uint8_t, int, int, OLEDDISPLAY_GEOMETRY, HW_I2C, int)':
C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SSD1306Wire.h:76:47: error: 'Wire1' was not declared in this scope
       this->_wire = (_i2cBus==I2C_ONE)?&Wire:&Wire1;
                                               ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

Anything to point me in the right direction would be great.  Thanks in advance.

Qrome, 2020-08-06:
What ESP8266 core do you have installed.   Make sure you have ESP8266 2.5.2 -- as mentioned in the readme.md file.
Also, make sure you have installed the listed required library files.

nogardvfx, 2020-08-06:
Yeah I followed all the instruction (or thought I have).  Grabbed the link for the .json file and put it in my preferences.  Grabbed the two zip files from the github for the wifi and oled and put them in Sketch->Include Library->Add .zip library.   Set the board to LOLIN(WEMOS) D1 R2 & Mini.  Set Flash Size to 4mb (FS:1MB OTA:~1019KB).  Bring in printmonitor.ino and hit the checkmark button.  

The hardware I purchased was:
https://www.amazon.ca/gp/product/B07PGKZBRL/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.ca/gp/product/B01N1IIX5J/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Anything else I can tell you?

Qrome, 2020-08-06:
Can you verify the ESP8266 Core of 2.5.3 or 2.5.2?  The fact that you are showing (FS:1MB OTA) tells me you probably have the wrong ESP8266 core installed.   It should say 4M (1M SPIFFS)

nogardvfx, 2020-08-07:
Yes I am a knucklehead and it was set incorrectly (now I have the correct 4m (1m SPIFFS)), but I continue to get the same error. I turned on verbose if that is helpful.

In file included from C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SH1106.h:33:0,

                 from C:\Users\nogardvfx\Downloads\printer-monitor-master\printer-monitor-master\printermonitor\printermonitor.ino:4:

C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SH1106Wire.h: In constructor 'SH1106Wire::SH1106Wire(uint8_t, uint8_t, uint8_t, OLEDDISPLAY_GEOMETRY, HW_I2C, int)':

C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SH1106Wire.h:74:47: error: 'Wire1' was not declared in this scope

       this->_wire = (_i2cBus==I2C_ONE)?&Wire:&Wire1;

                                               ^

In file included from C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SSD1306.h:33:0,

                 from C:\Users\nogardvfx\Downloads\printer-monitor-master\printer-monitor-master\printermonitor\printermonitor.ino:8:

C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SSD1306Wire.h: In constructor 'SSD1306Wire::SSD1306Wire(uint8_t, int, int, OLEDDISPLAY_GEOMETRY, HW_I2C, int)':

C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src/SSD1306Wire.h:76:47: error: 'Wire1' was not declared in this scope

       this->_wire = (_i2cBus==I2C_ONE)?&Wire:&Wire1;

                                               ^

Using library esp8266-oled-ssd1306-master at version 4.1.0 in folder: C:\Users\nogardvfx\Documents\Arduino\libraries\esp8266-oled-ssd1306-master 

Using library Wire at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\Wire 

Using library brzo_i2c-master at version 1.3.3 in folder: C:\Users\nogardvfx\Documents\Arduino\libraries\brzo_i2c-master 

Using library SPI at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SPI 

Using library WiFiManager-master at version 0.15.0 in folder: C:\Users\nogardvfx\Documents\Arduino\libraries\WiFiManager-master 

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi 

Using library ESP8266WebServer at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer 

Using library DNSServer at version 1.1.1 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\DNSServer 

Using library esp8266 at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\esp8266 

Using library ESP8266mDNS at version 1.2 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS 

Using library ArduinoOTA at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ArduinoOTA 

Using library ESP8266HTTPUpdateServer at version 1.0 in folder: C:\Users\nogardvfx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPUpdateServer 

exit status 1

Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

Qrome, 2020-08-07:
Make sure you have this library installed:  https://github.com/ThingPulse/esp8266-oled-ssd1306

nogardvfx, 2020-08-07:
Yeah I do have it already installed.

nogardvfx, 2020-08-07:
ok move from Windows to my wife's mac and everything compiled.

nogardvfx, 2020-08-07:
Another question, if you do not mind, since my screen is still black.  I can make a new topic if you wish.

I did not get the UCTRONICS 0.96 Inch OLED screen, but have kuman 0.96 Inch.
https://www.amazon.ca/gp/product/B01N1IIX5J/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

Should that make any difference?  I purchased 2 and they both are black.

Qrome, 2020-08-07:
Oh, glad you got it to compile.   Sounded like your other machine was still referencing different library files or could not find them.

No need to create a new thread.   

Question:  did you change anything in the settings.h file?   Looks like the one you got should work just fine.   Just make sure it is using the right pins to the Wemos d1 mini.   Is it connected like the diagram?

nogardvfx, 2020-08-07:
I did not change a things in setting.h.  I wouldn't even have dared.  I did match the wiring exactly.   Checked about 10 times thinking that was the case.  Attaching some pics just in case.
[image]
[image]
[image]

Apologies on my soldering skills.

nogardvfx, 2020-08-07:
Just some additional info, I have two boards and two screens.  I just tried this on the second board with the same results.

Qrome, 2020-08-07:
So, if it is only the OLED that isn't working, have you been able to connect to it via the AP hotspot when it boots up?   You should be able to configure it for wifi then pull up the web interface.   Also, have you opened the serial monitor when it is booting up with baud 115200 -- it might give you some information that is going on.   I hope we can get you working.

The Wemos you have is different the ones I use.  I wonder if it is booting up ok.

nogardvfx, 2020-08-07:
Its is a v3.  I have ordered the older ones and should get them on Sunday.  

Yes, I connected through the AP and set it up to my wifi network.  I was able to log in and everything worked (except the display).

Let me look into the serial interface.

wabbitguy, 2020-08-07:
I found some of those OLEDS use a different display address in the settings.h file...

const int I2C_DISPLAY_ADDRESS = 0x3c; // I2C Address of your Display (usually 0x3c or 0x3d)

Perhaps try 0x3d for the Kuman to make the OLED interface work:

const int I2C_DISPLAY_ADDRESS = 0x3d; // I2C Address of your Display (usually 0x3c or 0x3d)

Qrome, 2020-08-07:
Oh, good call.   Yes, try changing the I2C address for the display in the Settings.h file.

nogardvfx, 2020-08-08:
Tried that and no luck.  Much sadness.

Qrome, 2020-08-08:
For fun, have you tried swapping the green and yellow wires?

wabbitguy, 2020-08-08:
I just had a look at the link to the Amazon OLED display and the address is shown on the back of the display (there's a resistor to select that address). From the image on Amazon the address is 0x3c so the settings are correct.

I noticed that in the middle photo of the back of the WEMOS, the Yellow Dupont inside connector looks like it was installed upside down. I wonder if the male pin is pushing the female pin away from the connection? It doesn't look like the other connectors any way...just a guess.

nogardvfx, 2020-08-08:
I have two sets of boards and two sets of screens.  I originally wired the connectors together before deciding to redo the set up with the connectors.  I have tried multiple wires with the same results.  I even recognize those pictures were not the best and I did re-seat the connectors thinking they were not making a connection.  I have also tried swapping the green and yellow, but nothing on there.

I have ordered a set of the older boards (get those on Sunday).  I also ordered the exact model of the screen from the US, but it will take a few weeks to get up here to Canada.  I didn't want to take a chance on another brands of screen.

My feeling is too many variables right now. Luckily the parts are inexpensive.

Qrome, 2020-08-08:
@nogardvfx - sorry for all the frustration.  I honestly have yet to not have a 0.96" OLED I2C display not work for me and I have tried many of them.  I don't think it is your wemos -- I know many have used that one and I helped a guy a while back with one.

nogardvfx, 2020-08-08:
Ok will give it a try!

nogardvfx, 2020-08-08:
okay funny thing. I set up the pin switch and plugged the device in and the screen worked!  I didn't even recompile and upload any new code.

Qrome, 2020-08-09:
Well, that is awesome.   Glad it is working for you.

nogardvfx, 2020-08-09:
Yes thank you so much.  Just wanted to add one more thing.  I did do the switch to address 0x3d last night.  I never set it back.  Maybe helpful for future reference.