Qrome/printer-monitor #117: Username and Password account for the Web Interface problem.

MakerCrazySeb, 2020-09-26:
Hello,

I changed the Username and Password account for the Web Interface.
But, I must have made a typo and since then I can not connect.
I uploaded a new version to the card, but this did not reset the Username and Password.
I even uploaded another unrelated program and then reinstated PrintMonitor, but I'm still stuck.
Could you please give me the method to follow?
Thanks for your help and your great project.

MakerCrazySeb, 2020-11-05:
No one to help me, can't we make mistakes?

Qrome, 2020-11-05:
Sorry, just saw this.   Connect to the Arduino Serial Monitor using baud rate 115200 -- press the reset button and when it boots up it will display what the user / password is.   Login using that then disable it or change it as needed.

aristeu, 2021-05-10:
Thanks for this question/answer (and wow, the whole project and software are amazing, thanks for that).
What probably happened to Seb is what happened to me: typed a long password and it only saved part
of it (11 characters). My suggestion is javascript to prevent long passwords or a note about password rules.