Qrome/printer-monitor #120: Open Wireless Network "FaryLink"

mawo0030, 2020-11-22:
I successfully compiled the printer-monitor today, after setting up Wifi everything runs smooth, except that the ESP8266 spawns an unwanted open Network now, SSID "FaryLink", see screenshot:
[image]

I tried connecting to it, it has IP 192.168.4.1 and DHCP enabled as far as I can see.
There's a discussion on StackExchange [https://arduino.stackexchange.com/questions/47806/what-is-farylink-access-point](url) on the subject with a proper solution:

> after adding "WiFi.mode(WIFI_STA);" in the servers constructor, the mysterious FairyLink was removed. THX – clockw0rk Dec 16 '17 at 13:53

So I guess that line should be added somewhere in the printer-monitor.ino.

wabbitguy, 2020-11-22:
I have 4 PrintMonitors running and I don't see any of the FaryLink hot spots.

The code referenced as the fix is included in the WifiManger library for auto connect that is called from the SetUp in printermonitor.ino:

`if (!wifiManager.autoConnect((const char *)hostname.c_str())) {// new addition`

And in WifiManager.cpp 0.15.0 we find:

`
boolean WiFiManager::autoConnect(char const *apName, char const *apPassword) {
  DEBUG_WM(F(""));
  DEBUG_WM(F("AutoConnect"));

  // read eeprom for ssid and pass
  //String ssid = getSSID();
  //String pass = getPassword();

  // attempt to connect; should it fail, fall back to AP
  WiFi.mode(WIFI_STA);

  if (connectWifi("", "") == WL_CONNECTED)   {
    DEBUG_WM(F("IP Address:"));
    DEBUG_WM(WiFi.localIP());
    //connected
    return true;
  }

  return startConfigPortal(apName, apPassword);
}
`

Maybe some of the libraries are out of date that are being used.

Qrome, 2020-11-22:
I agree, this is not an issue related to this project at all.

closed 2020-11-22 by Qrome