Qrome/printer-monitor #122: Open Weather Map - NOT active ==> Serial monitor says "Unexpected response:"

CSauter, 2020-12-05:
Dear audience,
printer-monitor says: Please configure weather API.
I did this now several times, no change.
The error stays since creating my printer-monitor around 6 weeks ago
==> All of the rest of the printer-monitor is working fine without any problems

**1) WeatherApiKey is verified and working**

**2) Serial monitor says after "saving the configuration"**
_IP addresses are masked with xxx.xx
Octoprint-API & Accounts are masked with  ----_
```
---snip---
15:46:19.675 -> Saving settings now...
15:46:19.708 -> UtcOffset=1.00
15:46:19.708 -> PrinterApiKey=----------------------------
15:46:19.708 -> PrinterHostName=octopi
15:46:19.708 -> PrinterServer=192.168.xxx.xx
15:46:19.708 -> PrinterPort=80
15:46:19.708 -> PrinterName=
15:46:19.708 -> PrinterAuthUser=------
15:46:19.708 -> PrinterAuthPass=--------
15:46:19.708 -> minutesBetweenDataRefresh=15
15:46:19.708 -> themeColor=red
15:46:19.743 -> IS_BASIC_AUTH=1
15:46:19.743 -> www_username=-------
15:46:19.743 -> www_password=
15:46:19.743 -> DISPLAYCLOCK=1
15:46:19.743 -> IS_24HOUR=1
15:46:19.743 -> INVERT_DISPLAY=0
15:46:19.743 -> USE_FLASH=1
15:46:19.743 -> DISPLAYWEATHER=1
15:46:19.743 -> WeatherApiKey=1b446c7ca83894162e2ab21a5dd6dadf
15:46:19.743 -> CityID: 2867714
15:46:19.743 -> IS_METRIC=1
15:46:19.743 -> WeatherLanguage=de
15:46:19.743 -> HAS_PSU=0
15:46:19.778 -> 
15:46:19.778 -> Getting Weather Data...
15:46:19.778 -> Getting Weather Data
15:46:19.778 -> GET /data/2.5/group?id=2867714&units=metric&cnt=1&APPID=1b446c7ca83894162e2ab21a5dd6dadf&lang=de HTTP/1.1
15:47:19.723 -> Waiting for data
15:47:24.737 -> Response Header: 

15:47:24.737 -> Unexpected response:

15:47:24.737 -> Updating Time...
15:47:25.795 -> .
15:47:25.795 -> 14:48:23
15:47:25.795 -> 14:48:23
15:47:25.795 -> 53303
15:47:30.788 -> Local time: 3:48 PM
15:47:30.890 -> Signal Strength (RSSI): 92%
15:47:32.750 -> Getting Octoprint Data via GET
15:47:32.785 -> GET /api/job HTTP/1.1
15:47:32.820 -> Status: Printing
---snap---
```

**3) The _manual_ assembled URL will look like this:**
http://api.openweathermap.org/data/2.5/group?id=2867714&units=metric&cnt=1&APPID=1b446c7ca83894162e2ab21a5dd6dadf&lang=de

**4) With the seemingly successful result for my desired city:**
{"cnt":1,"list":[{"coord":{"lon":11.58,"lat":48.14},"sys":{"country":"DE","timezone":3600,"sunrise":1607150895,"sunset":1607181658},"weather":[{"id":500,"main":"Rain","description":"Leichter Regen","icon":"10d"}],"main":{"temp":2.9,"feels_like":0.4,"temp_min":2.78,"temp_max":3.33,"pressure":1000,"humidity":93},"visibility":50,"wind":{"speed":1.16,"deg":66},"clouds":{"all":100},"dt":1607177336,"id":2867714,"name":"Munich"}]}

**Result:**
This makes me think that something is not working in the code?

Thank you in advance

Qrome, 2020-12-05:
Yeah, your API key is valid and pulls data.  It looks like the device or device IP is restricted on making external calls to the public internet.  Could this be the case?  Do you require a proxy on your network?

Quite sure there isn't a problem with the code as this project has been running for years now.

Note this section in the log:
15:47:19.723 -> Waiting for data
15:47:24.737 -> Response Header: 

15:47:24.737 -> Unexpected response:

The device is not getting a valid response.

Tragl1968, 2020-12-06:
Hi Qrome,
yes, mea culpa.
The issue is solved.

I used my standard profile of Internet-Access-filters on my Firewall. Every device runs fine with it.
My filters exclude content harmful to minors and website access via IP addresses.
If I deactivate this filter, the printer monitor works. ufff ...

Thanks for pointing out the obvious ... ;-)
As I don't expect any harmful content to minors I assume that there is a call to an IP-address while handshaking with API.openweather.

Again, thank you for your fast answer, this is really cool support.

closed 2020-12-06 by Tragl1968