Qrome/printer-monitor #128: Wifi APCallback ESP

denkyem, 2021-02-11:
After a fresh install or "forget Wifi", I connect to the SSID "PrintMon-xxxxxx" configure my wireless network and everything works fine.

Then if I disconnect and connect again the power, the esp connect correctly to my wireless network and everything works great, but also creates a ESP-xxxxxx network, that I can connect and have access to the UI.

soundylinz, 2021-04-08:
I have exactly the same problem.

If it has to do with the WifiManager, then unfortunately I have not yet found the correct information on how to deactivate this function.

Is there anything?

Qrome, 2021-04-08:
So you are saying it is still creating a wifi hotspot after it connects and gets an IP from your DHCP server?   None of my 8 here in my home do ping the ESP-xxxxxx -- does it resolve the same IP as your DHCP server.

soundylinz, 2021-04-08:
I am successfully connected to the WLAN, get an IP (192.168.1.113) and can also access the printer monitor via the IP.

In addition, an AP with the IP 192.168.4.1 is started. As an open WLAN!

Everyone in my area can log in! I don't like that at all and it's a security problem, isn't it?

soundylinz, 2021-04-08:
I do...

1. Reset Settings
2. Forget Wifi

Setup WIFI once agin...

The problem is solved, but why... *confused*

soundylinz, 2021-04-08:
But, when i flash it with Arduino again (without reset or forget wifi) the AP "ESP-xxx" is once again here and open. :-(