Qrome/printer-monitor #129: Printer monitor percent complete does not match OctoPrint interface

chicago925, 2021-02-12:
The percent completion on the Printer Monitor screen does not match the OctoPrint percent complete shown on the web page. In addition it does not match the Cura Monitor percent complete (which does match the OctoPrint interface. The Printer Monitor percent completion is always about 4 - 5% behind the OctoPrint percent completion.

Print time remaining and print time completed is correct, so is the calculation based on an API call or a comparison of these two numbers? In any case it is incorrect.

cchaz003, 2021-02-27:
I also have this issue. It doesn't hurt the experience too much but I am curious about why this is. I would have figured that the percentage is just pulled from the octoprint api directly, no?

cchaz003, 2021-02-27:
Well I just answered my own question. It is pulling the progress straight from the API but for some reason the API looks like it's giving a differing answer than the Octoprint web UI. I do have the PrintTimeGenius plugin installed. I wonder whether that not only changes the estimated print time but also changes the web UI percentage complete... Either way at least in my case it doesnt appear to be an issue stemming from the printer-monitor side of things.