Qrome/printer-monitor #132: Screen not working

smokey7722, 2021-02-26:
I purchased the components recommended, programmed the unit and wired as directed but the screen won't display anything.  When the display is plugged in the ardiuno powers up and the blue led on it turns on solid, then blinks 5 times then goes out.  Then it repeats shortly thereafter.  If I have the screen unplugged, the ardiuno powers up without those led blinks and it connects to wifi for me to access.  With the screen it doesn't seem to be powering up as it isn't connecting to WiFi.

I have replaced the screen thinking it was bad and tried various other power supplies just to be safe.

Hoping for some assistance as I haven't been able to figure this out at this point.

Qrome, 2021-02-26:
What pins did you connect the display to on the Wemos D1 Mini?

smokey7722, 2021-02-26:
The standard D2 / D5 as recommended.  I did try various versions of the ESP8266 package too (originally the newest (2.7.4) then downgraded to 2.4.1 and it didn't seem to make a difference.  I'm new to Arduino though so not sure if its just a matter of re-uploading the sketch after that change in the IDE or if there is an extra step too?

Qrome, 2021-02-26:
Use esp8266 Core platform version 2.5.2 -- this is stated in the readme.   I would confirm your wired up correctly or that the display is working correctly.   Are you using the larger display or the smaller one?

smokey7722, 2021-02-26:
I'm on the smaller display right now (bought a second different one as well but either way its the smaller one) and nothing on that.  I was playing with different versions based on the other "my screen doesn't work" posts.  Changing to 2.5.2 now though to check that one.

smokey7722, 2021-02-26:
Looks like similar behavior.  When powered up, blue led on the Wemos turns on for about 8 seconds, then blinks 5 times, then goes back to on for a few seconds then off.  Nothing on the screen though.  I have verified the wiring for the screen numerous times as well as I figured I just screwed that up but that looks good.

Qrome, 2021-02-26:
The blue led on the wemos is an activity light.   That indicates that it is booting up and starting.   What do you see in the Serial Monitor using Baud 115200 on the startup?

smokey7722, 2021-02-26:
Yup 1M SPIFFS is setup.  My RPI is off right now so it won't connect to it (looks like I may have reset things so the config isn't there to connect to it anyway but the weather stuff is in there and pulling though) and it does come on the WiFi.

Startup with serial monitor:

```
13:34:15.162 -> UtcOffset=-8.00
13:34:15.162 -> PrinterApiKey=
13:34:15.162 -> PrinterHostName=octopi
13:34:15.162 -> PrinterServer=
13:34:15.162 -> PrinterPort=80
13:34:15.162 -> PrinterName=
13:34:15.162 -> PrinterAuthUser=
13:34:15.162 -> PrinterAuthPass=
13:34:15.162 -> minutesBetweenDataRefresh=15
13:34:15.197 -> themeColor=light-green
13:34:15.197 -> IS_BASIC_AUTH=1
13:34:15.197 -> www_username=admin
13:34:15.197 -> www_password=password
13:34:15.197 -> DISPLAYCLOCK=1
13:34:15.197 -> IS_24HOUR=0
13:34:15.197 -> INVERT_DISPLAY=0
13:34:15.197 -> USE_FLASH=1
13:34:15.197 -> DISPLAYWEATHER=1
13:34:15.197 -> WeatherApiKey=fa3d16e529380d272c69362d76d91b90
13:34:15.197 -> CityID: 5506956
13:34:15.197 -> IS_METRIC=0
13:34:15.197 -> WeatherLanguage=en
13:34:15.197 -> HAS_PSU=0
13:34:15.197 -> *WM: [1] AutoConnect 
13:34:15.197 -> *WM: [2] Connecting as wifi client... 
13:34:15.232 -> *WM: [3] STA static IP:
13:34:15.232 -> *WM: [2] setSTAConfig static ip not set, skipping 
13:34:15.232 -> *WM: [1] Connecting to SAVED AP: *******
13:34:15.232 -> *WM: [3] Using Password: *********
13:34:15.232 -> *WM: [3] WiFi_enableSTA enable
13:34:15.232 -> *WM: [3] enableSTA PERSISTENT ON 
13:34:15.721 -> *WM: [3] Mode after delay:  STA
13:34:15.721 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
13:34:20.846 -> *WM: [2] Connection result: WL_CONNECTED
13:34:20.846 -> *WM: [1] AutoConnect: SUCCESS 
13:34:20.846 -> *WM: [1] STA IP Address: 10.0.0.87
13:34:20.846 -> Signal Strength (RSSI): 68%
13:34:20.846 -> Server started
13:34:20.846 -> Use this URL : http://10.0.0.87:80/
13:34:23.663 -> no services found - make sure Printer server is turned on
13:34:23.663 -> *** Leaving setup()
13:34:23.663 -> *WM: [3] unloading 
13:34:23.663 -> 
13:34:23.663 -> Getting Weather Data...
13:34:23.663 -> Getting Weather Data
13:34:23.663 -> GET /data/2.5/group?id=5506956&units=imperial&cnt=1&APPID=********&lang=en HTTP/1.1
13:34:25.554 -> Waiting for data
13:34:25.554 -> Response Header: HTTP/1.1 200 OK
13:34:25.554 -> lat: 36.175
13:34:25.554 -> lon: -115.1372
13:34:25.554 -> dt: 1614373819
13:34:25.554 -> city: Las Vegas
13:34:25.554 -> country: US
13:34:25.554 -> temp: 60.89
13:34:25.554 -> humidity: 8
13:34:25.554 -> condition: Clear
13:34:25.554 -> wind: 5.75
13:34:25.554 -> weatherId: 800
13:34:25.554 -> description: clear sky
13:34:25.554 -> icon: 01d
13:34:25.554 -> 
13:34:25.554 -> Updating Time...
13:34:26.575 -> .
13:34:26.575 -> 21:34:25
13:34:26.575 -> 21:34:25
13:34:26.575 -> 77665
13:34:31.607 -> Local time: 1:34 PM
13:34:31.607 -> Clock Mode is turned on.
13:34:37.295 -> Signal Strength (RSSI): 58%

```

Qrome, 2021-02-26:
Yeah, looks like you have it all working on the Wemos D1 Mini.  Looks like an issue with your display.   This is an I2C display, right?   It only has 4 wires on the back.  Have you tried switching the D2 and D5 pins?

smokey7722, 2021-02-26:
I've tried these two displays.  Each only had 4 wires and I took note of the pinout on the displays to be sure to wire them right.

https://www.amazon.com/gp/product/B00O2LKEW2/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B072Q2X2LL/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

Are you referring to switching the SCL from D5 to D2 and SDA from D2 to D5 by switching them?  Or are you referring to changing the code to map to a different pin (and move the wire on the Wemos to the new pin) to test?

Qrome, 2021-02-26:
I know it is a long shot, but there have been times where I have seen the pins labeled wrong.
The other option is to change the SCL pin to D1 and update the Settings.h file line 95 to D1.
Find:
const int SCL_PIN = D5; // original code D5 -- Monitor Easy Board use D1
Change:
const int SCL_PIN = D1; // original code D5 -- Monitor Easy Board use D1

smokey7722, 2021-02-26:
I tried swapping and it didn't help.  Also just changed D5 to D1 and moved SCL wire to it and didn't make any difference unfortunately :(

Qrome, 2021-02-27:
Wow, well, isn't from a lack of trying.   Could be one of two things, bad board or the 1306 display library is not correct.   You can try loading one of the compiled bin files provided.     Feel free to ping me on thingiverse messages and I might have another option for you.

DuschdrBabbe, 2021-05-06:
Try to connect 3V to the display
had also this prob...