Qrome/printer-monitor #133: When I flip display orientation, it flips but the colors don't.

jurassic73, 2021-03-05:
When I flip display orientation on my blue/yellow oled, the contents flip but not the color scheme so I end up with half yellow, half blue clouds for example.

Thanks for this great tool!  Stoked to build one.

Qrome, 2021-03-05:
That section of the screen is always yellow -- it is not controlled by software.   If you want the yellow at the bottom, you will have to invert the display.  The actual OLED is yellow.

closed 2021-03-05 by Qrome

jurassic73, 2021-03-05:
Ah okay - thanks for this.  Didn't realize.

Thanks for the great build.  Nice addition to my printer setup.

jurassic73, 2021-03-06:
Have another question though... any option to reflect daylight savings automatically(daylight savings is a pita :) )?

Also, any chance you could add an Estimated Time for print completion?