Qrome/printer-monitor #134: Enhancement: Display Layer Progress Integration

Dougsterde, 2021-03-07:
Would love to see this integrated with the "Display Layer Progress" plugin.

Would also love to see "Time When Finished"

victor7376, 2021-03-13:
Hopefully Qrome will add them in as I’ve just forked this and added them both in.

Dougsterde, 2021-03-19:
Just got a chance to check out your fork of Qrome/printer-monitor.

Love it!  Fixed the font size for the layer display.

Other than that…awesome!

Great work.

Doug.

victor7376, 2021-03-19:
Can I ask what font size you used, as I put it as a smaller font due to some maybe having over 1000 layers as I wasnt sure how it would look or fit.

Glad you like it btw.

Dougsterde, 2021-03-20:
I just used the same as the other screens. 14 and 16 I believe. I haven't printed anything with over 1000 layers. Didn't think of that. Could add code to reduce the font if needed.

victor7376, 2021-03-20:
I've just updated the code to switch between the different layer amounts, so if its 1000 or more it makes the text smaller otherwise its the same as all the other screens.

I checked on the font size and its 16 & 24, so if its equal or greater than 1000 layers it'll put the text to 16 instead of 24

Dougsterde, 2021-03-21:
I get it downloaded and check it out.

Dougsterde, 2021-06-16:
Are you seeing odd time on End Time screen?  Mine always show 11:25

victor7376, 2021-06-17:
Any chance you can send a photo?

Dougsterde, 2021-06-17:
Next time I print i will. Not sure if it is issue with printtimegenius or printer-monitor. Shows remaining time of about an hour then ETA about 6 hours from current. Going to try to remove and reinstall printtimegenius and see if it makes a difference.

morphias2004, 2023-04-22:
I'm getting this after uploading the code from either @victor7376

[image]

and this with the code from @Dougsterde . :(

[image]

I'm guessing library version of SSD1306 are the issue?

If I upload the original code from Qrome, and use SSD1306 v4.1.0 by ThingPulse I have no issue.

Any assistance is greatly appreciated.

morphias2004, 2023-04-22:
Leaving the above comment for the benefit of others.

The issue was that in @victor7376 and @Dougsterde code, I had forgotten to uncomment `#define DISPLAY_SH1106` in the Settings.h as I'm using the 1.3" SSH1106 display.

SSD1306 by ThingPulse v4.1.0 library is still required or the display will not work.

[image]