Qrome/printer-monitor #135: Added Features

victor7376, 2021-03-13:
Hi Qrome,

I've just made a branch of this and added 2 new features:

1, screen for current Layer
2, screen for estimated end time (clock)

But I've left them out of the RepetierClient as I dont use that so unable to test it.

The info has been pulled from the 'DisplayLayerProgress' plugin.

victor7376, 2021-03-13:
For anyone thats cloning the code, I've just added the Layer progress, Estimated Time into the webpage & added Next Update.

markstamp, 2021-04-08:
Can you let me know how i clone and update the code for printmonitor to get this to work?

victor7376, 2021-04-08:
> Can you let me know how i clone and update the code for printmonitor to get this to work?

You can make your own fork, then download the code from my fork and paste into your own fork.

There might be a better way of doing it if someone knows.

DuschdrBabbe, 2021-05-10:
can you also add the eta to the display not only to the webpage?
please also but then the new bin file in your repo

thank you!!!!

Greetings from Germany

Pascal

victor7376, 2021-05-10:
I've just uploaded a bin file which has been compiled from the features.

ETA - both the actual time when the print is finished & how many hours / minutes till print is finished is programmed in.

DuschdrBabbe, 2021-05-11:
Super thanks... just downloaded your bin and fished to the Esplanade... Layer and Time are now a part of the run, but no values are displayed ... I have installed the display eta plugin in octoprint... 

also on the webpage it displays:

Status: Offline
Reason: Response: HTTP/1.1 404 NOT FOUND

but also:
File: dragon_pikachu_0.4n_0.2mm_PLA_I3MEGAS_6h41m.gcode
File Size: 33059.35KB
Filament: 18.82m
Layer: / 
Estimated Finish Time: 
Tool Temperature: 210.0° C
Bed Temperature: 60.0° C
Est. Print Time Left: 06:17:58
Printing Time: 00:17:13

a mix of offline and printings values... 

the test of the api shows me

`{"job":{"averagePrintTime":null,"estimatedPrintTime":23701.338352307735,"filament":{"tool0":{"length":18820.52734375,"volume":45.268673305480156}},"file":{"date":1620633706,"display":"dragon_pikachu_0.4n_0.2mm_PLA_I3MEGAS_6h41m.gcode","name":"dragon_pikachu_0.4n_0.2mm_PLA_I3MEGAS_6h41m.gcode","origin":"local","path":"dragon_pikachu_0.4n_0.2mm_PLA_I3MEGAS_6h41m.gcode","size":33852777},"lastPrintTime":null,"user":"pascal"},"progress":{"completion":2.3394890174002563,"filepos":791982,"printTime":1089,"printTimeLeft":22633,"printTimeLeftOrigin":"genius"},"state":"Printing"}` 

greetings

Pascal

victor7376, 2021-05-11:
It requires the ‘Display Layer Progress’ plugin to be installed.  To check if you have the plug-in installed go to the printers IP with the following added on ‘/plugin/DisplayLayerProgress/values’.  So if your printers ip is 192.168.0.50 for example you would go to:

http://192.168.0.50/plugin/DisplayLayerProgress/values

I would restarted your monitor and printer to make sure everything works.

HappySl0th, 2021-07-19:
> Hi Qrome,
> 
> I've just made a branch of this and added 2 new features:
> 
> 1, screen for current Layer
> 2, screen for estimated end time (clock)
> 
> But I've left them out of the RepetierClient as I dont use that so unable to test it.
> 
> The info has been pulled from the 'DisplayLayerProgress' plugin.

Since there is a clock function, are you able to add in a feature which turns off the clock between set time?

victor7376, 2021-07-19:
> Since there is a clock function, are you able to add in a feature which turns off the clock between set time?

Can you explain a little better please?

HappySl0th, 2021-07-19:
Currently there is only the option to display the clock and weather 24/7. A good addition is an option to choose to show the clock between 9am to 5pm for example, and the other hours the display will be off.

victor7376, 2021-07-19:
> Currently there is only the option to display the clock and weather 24/7. A good addition is an option to choose to show the clock between 9am to 5pm for example, and the other hours the display will be off.

ok, understood.  possible to be imported from the marquee scroller which has that feature

victor7376, 2021-07-19:
> Currently there is only the option to display the clock and weather 24/7. A good addition is an option to choose to show the clock between 9am to 5pm for example, and the other hours the display will be off.

I haven't tested, but this does compile.  Coded in the settings section:

```
String timeDisplayTurnsOn = "06:30";  // 24 Hour Format HH:MM -- Leave blank for always on. (ie 05:30)
String timeDisplayTurnsOff = "23:00"; // 24 Hour Format HH:MM -- Leave blank for always on. Both must be set to work.
```

and in the printer.ino:

```
void checkDisplayPower() {
  if (timeDisplayTurnsOn == "" || timeDisplayTurnsOff == "") {
    return; // nothing to do
  }
  String currentTime = timeClient.getHours() + ":" + timeClient.getMinutes();;

  if (currentTime == timeDisplayTurnsOn && !displayOn) {
    Serial.println("Time to turn display on: " + currentTime);
    display.displayOn();
  }

  if (currentTime == timeDisplayTurnsOff && displayOn) {
    Serial.println("Time to turn display off: " + currentTime);
    display.displayOff();
  }
}
```

Dont forget to add:

`checkDisplayPower();`

into the Loop section & handleUpdateWeather section.  I'm trying to cover all bases.  The loop section should be enough.

HappySl0th, 2021-08-04:
@victor7376  I added checkDisplayPower as above and did not remove checkDisplay and added checkDispalyPower to the loop and it compiled ok but it isn't working on the display end. The screen is still on after hours. I had a look at the marquee scroller code and noticed that the author only have checkDisplay with the timeDisplayTurnOn code inside. Do you have any idea what's wrong? Thanks

victor7376, 2021-08-04:
> @victor7376 I added checkDisplayPower as above and did not remove checkDisplay and added checkDispalyPower to the loop and it compiled ok but it isn't working on the display end. The screen is still [...]

I’ll take a look at the code hopefully today.

victor7376, 2021-08-04:
> @victor7376 I added checkDisplayPower as above and did not remove checkDisplay and added checkDispalyPower to the loop and it compiled ok but it isn't working on the display end. The screen is still [...]

Done some quick testing and I've added the 'checkDisplayPower' in 3 different places, its allowing the unit to turn off at the correct time, but some reason its not turning the unit back on at the stated time.  The only other possible way would be to try and program something into the unit so that when you visit the clocks server, you click on a few links which would turn the clock on and off.  I'll take a look at this option tomorrow.