Qrome/printer-monitor #138: won't connect to octoprint, always says "OFFLINE"

geneb389, 2021-04-21:
First off, this is really awesome. It worked initially, but I shut off the octoprint, and then when I turned it back on the print monitor would not reconnect and just stayed offline. I have tried reflashing the firmware on the print monitor, Verified the settings on the print monitor. I can still connect to the octoprint via the App and web interface but the print monitor just displays the weather and time but says "OFFLINE"

Qrome, 2021-04-21:
Make sure the address to your octoprint server is still accurate.   There is also a test button in the configuration.  Also, are you running a release candidate of OctoPrint or the stable version?

https://youtu.be/kcBspqWhpIU?t=365

geneb389, 2021-04-21:
!
[image]
[image]

i did verify the address. when i do the test. if it's not printing everything says NULL but operational, When printing it displalys what's in the picture, but the print monitor always displays OFFLINE

Qrome, 2021-04-21:
When did this start happening.   Also, can you tell me what version of OctoPrint you are running?   Is it a release candidate?

geneb389, 2021-04-21:
OctoPrint Version is 1.5.3

It started happening after I turned off the raspberry pi4 for a couple days. when I turned it back on, the print monitor wouldn't re-connect. The ip address and everything stayed the same. I first tried restarting the print monitor but that didn't work either. it is weird to me because the test button reports information, but the screen doesn't and shows it to be offline.

Qrome, 2021-04-21:
Is your pi4 and monitor on the same subnet (same IP range) where they can see each other?
Connect your Wemos D1 mini to your computer over the USB and open the Arduino IDE.  Open the Serial Port monitor using baud rate 115200 and then press the reset button on the side of the Wemos D1 mini.   Watch the serial monitor for the calls to the OcotoPrint server and see what you are getting there.

geneb389, 2021-04-22:
this is what the serial port monitor returned:

UtcOffset=-7.00
PrinterApiKey=A56FD75D00FB4BD5B25DF1B37D9FC957
PrinterHostName=PRUSA MMU2s
PrinterServer=192.168.1.24
PrinterPort=80
PrinterName=
PrinterAuthUser=prusa
PrinterAuthPass=prusa
minutesBetweenDataRefresh=10
themeColor=orange
IS_BASIC_AUTH=1
www_username=admin
www_password=password
DISPLAYCLOCK=1
IS_24HOUR=0
INVERT_DISPLAY=0
USE_FLASH=1
DISPLAYWEATHER=1
WeatherApiKey=fffad25e28e56b5297dd0e285ca12c7e
CityID: 5782055
IS_METRIC=0
WeatherLanguage=en
HAS_PSU=0
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 6
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.10
Signal Strength (RSSI): 100%
Server started
Use this URL : http://192.168.1.10:80/
no services found - make sure Printer server is turned on
*** Leaving setup()
*WM: freeing allocated params!

Getting Weather Data...
Getting Weather Data
GET /data/2.5/group?id=5782055&units=imperial&cnt=1&APPID=fffad25e28e56b5297dd0e285ca12c7e&lang=en HTTP/1.1
Waiting for data
Response Header: HTTP/1.1 200 OK
lat: 40.6377
lon: -112.2961
dt: 1619050104
city: Stansbury park
country: US
temp: 53.82
humidity: 50
condition: Clouds
wind: 11.5
weatherId: 802
description: scattered clouds
icon: 03d

Updating Time...
.
00:26:56
0:26:56
1616
Local time: 5:27 PM
Getting Octoprint Data via GET
GET /api/job HTTP/1.1
Connection to OctoPrint failed: 192.168.1.24:80

Clock Mode is turned on.
Getting Octoprint Data via GET
GET /api/job HTTP/1.1
Connection to OctoPrint failed: 192.168.1.24:80

Qrome, 2021-04-22:
Yep, connection issue to 192.168.1.24 --  is there an active firewall on your network.  This is probably an issue with a setting on your router.   The device can access the internet but not another IP on your network.  I have seen this before.

geneb389, 2021-04-22:
no firewall that I am aware of. any idea what setting I should check?

Qrome, 2021-04-22:
I don't know what router you have, but look for something like this:
[image]

geneb389, 2021-04-22:
That totally fixed the issue. thank you very much.

geneb389, 2021-04-22:
left you a donation, hope it helps with your projects.

Qrome, 2021-04-22:
Awesome!   Thank you so much.  Glad you got it working.

closed 2021-04-22 by Qrome