Qrome/printer-monitor #139: Support for Mainsailos

wernju1, 2021-05-02:
Hello, is it possible to create support for mainsailos?
Regards
Jürgen

victor7376, 2021-05-06:
If there’s a API which outputs json it should be possible to add it to this.  I’ve downloaded it to see if I can test it tomorrow.

victor7376, 2021-05-07:
I've had a look today and between the aps etc it does output an api & also emulates OctoPrint api.  I havent tested it yet, but just running Printer Monitor and putting in the IP of your mainsailos it should connect straight up to the OctoPrint Emulator to gather the information it requires.

Hope this helps.

wernju1, 2021-05-07:
Ok, i will give it a try.
Thanks for your support

eisthiele, 2022-02-06:
Unfortunately it does not work. Only bed and tool temperatures are displayed. Just like in the FluidOS support issue. there is also a 3D print monitor moonraker but unfortunately the screen is too big for me.

victor7376, 2022-02-06:
> Unfortunately it does not work. Only bed and tool temperatures are displayed. Just like in the FluidOS support issue. there is also a 3D print monitor moonraker but unfortunately the screen is too big for me.

If it’s from the guy who made a fork of Qromes version, you should be able to use the same screen from qromes work for that version too.

eisthiele, 2022-02-07:
> If it’s from the guy who made a fork of Qromes version, you should be able to use the same screen from qromes work for that version too.

in the issue where fluidOS is concerned, it doesn't work either! only the temperature is displayed.
there is another printer monitor that also works with moonraker/mainsail maybe the developer from here can take a look at it maybe he can see the solution! I would like to use this monitor.

vKnmnn, 2022-02-18:
I have added support for moonraker in pr #160

suxmcgee, 2024-01-31:
@vKnmnn so, stupid question here, how do I get my existing octoprint monitor to use your pr with moonraker support?

victor7376, 2024-01-31:
> @vKnmnn so, stupid question here, how do I get my existing octoprint monitor to use your pr with moonraker support?

Go to the #160 link click on the files tab and copy all the details / files where he’s made changes then program them into the D1 mini