Qrome/printer-monitor #140: Even with correctly set language the weather info is still in English

micheld93, 2021-05-22:
Hi, 

first, this is a very nice thing of technique. It runs very good but I have the problem that the weather forecast is shown in English even when I set "de" in the settings. The web interface shows "Clouds (Bedeckt)", the word in () is the German word. The OLED display still shows "Clouds". I cannot find the reason, so I hope there can is someone who can help?

Wishes

victor7376, 2021-05-23:
I think I’ve found the issue, but will have to test it to confirm.

victor7376, 2021-05-23:
After looking through the coding and the API, the printer monitor takes the correct details from the api and the api details is the issue for the language.

The 'main' section within the api is in english but the description is in the set language, so you could change the coding within the printermonitor.ino on line 991:

` display->drawString(0 + x, 24 + y, weatherClient.getCondition(0));`

changing to

` display->drawString(0 + x, 24 + y, weatherClient.getDescription(0));`

which should now change to your choice of language.  As I have said the issue is with the api not displaying the 'condition' in the chosen language.

micheld93, 2021-05-23:
@victor7376 Thanks for the detailed answer. Yes, I get the written description in my language. Thanks for the code.

closed 2021-05-23 by micheld93